markwal/OctoPrint-GPX #1: Possible to install on plain raspberian with Octoprint?

beniroquai, 2015-04-08:
Hello,

awesome work! I really hope I can drive my CTC Dual with your Code.
I've a raspbian with Octoprint installed. I followed the steps suggested on the Octoprint-website.
```
Then I've tried to install your branch. After the first try I get this log:
Changing monitoring state from 'Offline' to 'Opening serial port'
Connecting to: /dev/ttyACM0
Connected to: Serial<id=0x235b8d0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: ??????
Send: M105
Recv: ??????
Changing monitoring state from 'Connecting' to 'Closed'
```

I've done the following after succesful installing the Octoprint:
```
cd ~
sudo service octoprint stop
mv OctoPrint OctoPrintSave
git clone -b devel https://github.com/markwal/OctoPrint
cd OctoPrint
 sudo python setup.py install

git clone https://github.com/markwal/OctoPrint-GPX
git submodule update --init
 sudo python setup.py install
```

Well of course, there is a mistake somewhere. There is no folder called oprint as you've mentioned. Any idea whats wrong? 
Thank you:)

markwal, 2015-04-09:
Yes, oprint is the VirtualEnv that the OctoPi image creates for OctoPrint and all of its dependencies to install into.  Since you ran your system python using sudo, it should have just installed all that to stuff to your system wide python library.

From your logs, it looks to me like the x3g protocol handler isn't kicking in.  Did you flip it to x3g mode (from GCode) on the GPX settings panel?

You should also have something like this earlier in your log:
2015-04-08 20:09:28,161 - octoprint.plugin.core - INFO - 3 plugin(s) registered with the system:
|  CuraEngine (bundled) = /home/pi/OctoPrint/src/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/src/octoprint/plugins/discovery
|  GPX (0.1) = /home/pi/OctoPrint-GPX/octoprint_GPX

And then when you click the connect button:
2015-04-08 20:10:54,199 - octoprint.util.comm - INFO - _openSerial
2015-04-08 20:10:54,210 - octoprint.plugins.GPX - INFO - Connecting through x3g.
2015-04-08 20:10:54,213 - octoprint.plugins.GPX - INFO - calling constructor /dev/ttyACM0 115200
2015-04-08 20:10:54,228 - GPX.gpxprinter - INFO - GPXPrinter created, port: /dev/ttyACM0, baudrate: 115200

beniroquai, 2015-04-09:
Awesome. That helped! It works. Can't believe, I'm controlling the Makerbot Clone via the Web Interface! Thank you so much!!

closed 2015-04-09 by beniroquai

t2ac32, 2017-01-27:
@markwal im adding my gpx.ini to cura plug in but it doesn't appear in . octoprint/plugins any ideas ?

markwal, 2017-01-27:
The cura plugin wants a cura .ini file (and particularly from version 15.04), not a gpx one.  You have to run cura on your desktop, get it configured to work right, test it via the gcode Upload, then export the settings to.ini file, then import to OctoPrint plugin.