markwal/OctoPrint-GPX #2: Getting UnicodeDecodeError every time I try to start a print

themaskedbear, 2015-05-15:
Hi,

First, thanks for the job you've done.

I have a makerbot replicator 2X with sailfish firmware.
I've generated a x3g file from makerware which I've uploaded to octoprint.
Every time I try to print the x3g uploaded I have this kind of error :

Error: UnicodeDecodeError: ''ascii' codec can't decode byte 0x9d in position 0: ordinal not in range(128)' @ comm.py:startPrint:528

And this for each x3g uploaded.

Strange thing is that putting the same x3g file on the SD card, and then launch the print from the octoprint interface works.

Do you have any clue about the problem ?

markwal, 2015-05-15:
Yes, OctoPrint doesn't understand x3g really. Only gcode, so to upload to OctoPrint and print from there, you need to use the gcode file. If you do though make sure the gpx.ini on the pi is what you want.

I'm working on making that simpler as well as a flashair plugin to allow OctoPrint to push the x3g to the SD card and start it from there.

markwal, 2015-06-23:
By the way, you can export the gcode from makerware and upload that instead.  In case that wasn't clear from my response.

closed 2015-06-23 by markwal