markwal/OctoPrint-GPX #3: can't start plugin (No PKG-INFO found for package: octoprint_GPX)

KipK, 2015-05-30:
I've followed your ReadMe file, but everytime I start Octoprint while GPX is installed I get this error. Haven't found yet how to fix this.

2015-05-30 09:19:09,944 - octoprint.server - INFO - Starting OctoPrint 1.2.0-dev-864-gceb0e29 (devel branch)
2015-05-30 09:19:09,953 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /usr/local/lib/python2.7/dist-packages/OctoPrint-1.2.0_dev_864_gceb0e29-py2.7.egg/octoprint/plugins and installed plugin packages...
2015-05-30 09:19:10,371 - py.warnings - WARNING - /usr/local/lib/python2.7/dist-packages/OctoPrint-1.2.0_dev_864_gceb0e29-py2.7.egg/octoprint/plugin/core.py:1096: UserWarning: No PKG-INFO found for package: octoprint_GPX
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2015-05-30 09:19:10,387 - octoprint.plugin.core - ERROR - Error loading plugin GPX
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.2.0_dev_864_gceb0e29-py2.7.egg/octoprint/plugin/core.py", line 582, in _import_plugin
    instance = imp.load_module(key, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint_GPX-v0.2_a1_6.g9731888-py2.7-linux-armv6l.egg/octoprint_GPX/__init__.py", line 10, in <module>
    from flask.exceptions import BadRequest
ImportError: No module named exceptions

KipK, 2015-05-30:
I haven't tested with binary build, provided link in the readme doesn't work. I've built GPX plugin from source directly.

KipK, 2015-05-30:
FInally I got it to work by manually installing it in plugins directory & installing GPX manually too.

setup.py develop was installing the plugin in the python/dist-package directory.

markwal, 2015-05-30:
It looks like you're installing the plugin to the global scope.  Are you using a raspberry pi and the OctoPi image?  If so, you want to make sure you activate the virtual environment first.  Installing to the plugins directory is a workaround that should work, but I haven't tested.

The reason that the scope might be a problem is that most of the plugins dependencies are installed with OctoPrint.

markwal, 2015-06-11:
@foosel spotted and fixed the issue you hit.  Your Flask is newer than mine and doesn't have "BadRequest" any more.

closed 2015-06-11 by markwal