markwal/OctoPrint-GPX #4: Can't Install Plugin

tonyphoang, 2015-06-01:
Hello, I'm relatively need to OctoPrint and I've attempted to install the GPX plugin to my newly formatted Octopi image. These were the steps I tried.

source ~/oprint/bin/activate
cd ~/OctoPrint
  git pull & git checkout devel
  python setup.py clean
  python setup.py install
  sudo service octoprint restart
cd ~
git clone https://github.com/markwal/OctoPrint-GPX
cd OctoPrint-GPX
git submodule update --init
python setup.py develop
mkdir ~/.octoprint/gpxProfiles
cp ~/OctoPrint-GPX/GPX/gpx.ini ~/.octoprint/gpxProfiles

But when I go onto http://octopi.local, select settings, I cannot find the GPX option. Can someone help?

markwal, 2015-06-02:
You'll want to restart OctoPrint and refresh your browser. I see you did that earlier, but you also need to do it after the plugin is installed.

sudo service octoprint restart

markwal, 2015-06-02:
That reminds me. Now that foosel includes the plugin manager bundled, I can update the instructions to use that to install instead.

tonyphoang, 2015-06-02:
I tried restarting the service and refreshing the browser with no success.

tonyphoang, 2015-06-02:
I'm at a total loss here. I've tried for a couple hours trying to dig through the internet for clues to get this to work but with no success. I'm trying to get a CTC 3d printer to work (Makerbot Rep 1 Dual/Flash Forge/ runs x3g)

markwal, 2015-06-02:
Could you upload your octoprint.log somewhere and put a link here?

It's in ~/.octoprint/logs

markwal, 2015-06-02:
You know, you may be hitting a recent bug in gpx.ini parsing. You could try to delete that gpx.ini and restart OctoPrint.

To get the fix:
cd ~/OctoPrint-GPX/GPX
git pull
cd ..
git pull
python setup.py install

Don't forget to activate the virtual env.

If you did hit my bug, I'm sorry about that.

tonyphoang, 2015-06-02:
I tried your fix and it did turn out to be a bug with the gpx.ini

The new codes I used were the following:
source ~/oprint/bin/activate
cd ~/OctoPrint
git pull & git checkout devel
python setup.py clean
python setup.py install
sudo service octoprint restart
cd ~
git clone https://github.com/markwal/OctoPrint-GPX
cd OctoPrint-GPX
git submodule update --init
python setup.py develop
mkdir ~/.octoprint/gpxProfiles
cp ~/OctoPrint-GPX/GPX/gpx.ini ~/.octoprint/gpxProfiles
cd ~/OctoPrint-GPX/GPX
git pull
cd ..
git pull
python setup.py install

The GPX setting now shows up in the setting.
When I try to connect to my printer, it gives me the following error.

Machine State: Error: Failed to open serial port, permissions correct?
File:  
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: -
Printed: -

The terminal shows:
Unexpected error while connecting to serial port: /dev/ttyACM0 IOError: 'AUTO port and baudrate not currently supported by GPX' @ comm.py:_openSerial:1268 (hook GPX)
Changing monitoring state from 'Offline' to 'Error: Failed to open serial port, permissions correct?'

markwal, 2015-06-02:
Ah good. You're back on track. Now just choose a specific port and baud rate (115200).

https://github.com/markwal/OctoPrint-GPX/issues/4#issuecomment-108051344

tonyphoang, 2015-06-02:
When I try to connect to a specific baudrate, I get this error:

Server is offline
The server appears to be offline, at least I'm not getting any response from it. I'll try to reconnect automatically over the next couple of minutes

So then I would have to get onto a terminal and run:
sudo service octoprint restart

in order for octoprint to run again.

tonyphoang, 2015-06-02:
Got it working!

tonyphoang, 2015-06-02:
Got it working!

markwal, 2015-06-02:
Wahoo!

tonyphoang, 2015-06-03:
Here is the way that I did it that worked for my CTC 3d printer. You can update your install procedures to reflect the current build.

Starting from a fresh install of octopi

source ~/oprint/bin/activate
cd ~/OctoPrint
git pull & git checkout devel
python setup.py clean
python setup.py install
sudo service octoprint restart
cd ~
git clone https://github.com/markwal/OctoPrint-GPX
cd OctoPrint-GPX
git submodule update --init
python setup.py develop
mkdir ~/.octoprint/gpxProfiles
cp ~/OctoPrint-GPX/GPX/gpx.ini ~/.octoprint/gpxProfiles
cd ~/OctoPrint-GPX/GPX
git pull
cd ..
git pull
python setup.py install
sudo service octoprint restart

open browser to http://octopi.local
login, select settings, select GPX, select enabled GPX, select makerbot for gcode flavor, select replicator 2 (default). Press ok.

Choose a specific port (usually only 1 is there - don't select AUTO) and baud rate (115200 or lower). Select connect.

The printer should work now.

markwal, 2015-06-03:
I'm glad you got it to work.  Sorry for the extra trouble.  Install update coming soon.

closed 2015-06-03 by markwal