markwal/OctoPrint-GPX #7: Can't update to Version 0.3.8

Helisinus, 2015-07-04:
Hi, 
today i did a fresh install on a new octopi installation. After this i had a installed version 0.3.7. This version works, but i get an update information for a new version 0.3.8. If i klick update, it seems the new Version will be installed and Octopi writes me i have to restart Octoprint to use the new version. But after the restart, i always have a version 0.3.7 and get the update message again.
Is there a other way to update this version?

Regards 
Joerg

markwal, 2015-07-04:
I think this may be a bug in how the plugin interacts with the softwareupdate plugin and pre-releases.  That is, I think this turns up when the most recent release is marked "pre-release", but since 0.3.7 was before I did the opt-in for pre-releases, it may be because the 0.3.7 plugin is answering that question wrong.

I've posted a new release 0.3.9 that contains several other fixes as well as the pre-release opt-in.  I just tried it out with softwareupdate and it appears to work.  This issue may come back the next time I post a pre-release update, but I'll be watching for it.

You may have to force the software update plugin to refresh its cache:

* Settings->Software Update->Advanced options->Force check for update (overrides cache)

Helisinus, 2015-07-04:
Sorry, but this didn't work for me. After Force check for update (overrides cache) i get the Message there ist a new version 0.3.9 available, and after installing this version, i got a message the update was successful and i have to restart octopring. But after restart octorprint or the hole raspberry the is still version 0.3.7 installed

Helisinus, 2015-07-04:
So, i have to correct my answer. In Software update the installed Version is shown as 0.3.7 and if i search for updates i will find every try the new Version. But in the Pluginmanager the Plugin was shown as version 0.3.9

markwal, 2015-07-05:
Hmmm...  The 0.3.7 local version in software update is stored in the software update plugin cache.  If you do that force check for update again (after the update), I think that will update the cache and it should match then the pluginmanager.

Ther purpose of that persisted cache is to avoid too many calls to the github API that is rate limited, but the cache also stores the local version.  Perhaps I'm doing something wrong in the plugin that prevents the software update plugin from updating its cache of local versions after an update is installed.

markwal, 2015-07-05:
What version of OctoPrint are you running?  It's in the lower left hand corner of the page.  Be sure to send the whole line including the parenthetical.

Helisinus, 2015-07-05:
There is only the Versionnumer Version: 1.2.2 (master branch) without parentheticat.
This is a new installation from yesterday an after the installation i did the updatet.
On my other (older) system is the thame after the update, only Version: 1.2.2 (master branch). no more information.
The system uses the version 0.3.9 but the updatedmanager thinks i use the version 0.3.7

Helisinus, 2015-07-05:
Here are a few screenshots:
[image]
[image]
[image]

markwal, 2015-07-05:
OK, this turns out to be fixed in the next version of the software update plugin.  A work around for now is to delete the the line plugins > softwareupdate > _config_version from config.yaml.

config.yaml is in ~/.octoprint

You're looking for the _config_version line in the softwareupdate section of the plugins section.

Helisinus, 2015-07-05:
o.k. this works. 
must i insert the line again when the update plugin is updatet or will the plugin insert this line ?

Thank for your fast help

markwal, 2015-07-05:
No. The software update plugin will recreate it. You might have to delete it again next time you run into this problem, but eventually you will get a new version of software update plugin that will fix it for good.

https://github.com/markwal/OctoPrint-GPX/issues/7#issuecomment-118615180

Helisinus, 2015-07-05:
o.k.
Thanks for your help :-)

closed 2015-07-05 by Helisinus

foosel, 2015-07-08:
For anyone else running into this problem, paste this

    [["remove", "plugins.softwareupdate._config_version", ""]]

into [the Yamlpatcher plugin](http://plugins.octoprint.org/plugins/yamlpatcher/).

[image]

Helisinus, 2015-07-08:
Sorry, but this dosen't work for me. Maybe i did something wrong.
See the attached Screenshots
[image]
[image]

foosel, 2015-07-08:
Don't close the settings dialog with save after you did that, just directy reboot. I'll release 1.2.3 probably tomorrow which should solve that issue for good though.

Helisinus, 2015-07-08:
I tryed the rebbot after pressing apply, But no success. Also if i delete the  line  _config_version  in config.yaml manualy. the update dont work. i have installed 0.3.9 and available is 0.3.10 and will not be installed. After a reboot the line _config_version was back in config.yaml.
But no matter, i will wait for the new Veriso 1.2.3 and try it again.

Thanks for your help

Joerg