markwal/OctoPrint-GPX #9: unable to upgrade from 3.13 to 3.15

ghvader, 2015-08-11:
When i tried to upgraded to the newly released 3.15 i am unable to
it thing it's 3.13
so it keeps telling me I have 3.13 but that there is an upgrade to 3.15
so net = unable to upgrade to newly released
I tried to remove and reinstall and i got 3.13 again.
Also 1 other minor issue
whenever i cancel a print from the printer, it disconnects me in octoprint and i have to re-connect.
is this normal? any way to not have to do that every time?
otherwise working great!!

markwal, 2015-08-11:
Oops.  Sorry about the update.  I uploaded the qualified name package and the software updater wants the version number only in the release folder.  Should be fixed now.

Yes, the cancel from the printer gets sent to OctoPrint as an error so it disconnects, which is the intended behavior at the current time.

ghvader, 2015-08-12:
that worked. thanks!

closed 2015-08-12 by ghvader