markwal/OctoPrint-GPX #11: Cant update from 0.3.16

giantpune, 2015-12-29:
I have a popup saying an update is available.  Current version is 0.3.16.  When I try to let it update, it always fails.   
```
2015-12-29 19:19:12,447 - octoprint.plugins.softwareupdate - ERROR - Update of gpx can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 626, in _perform_update
    update_result = updater.perform_update(target, check, target_version)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/plugins/softwareupdate/updaters/pip.py", line 60, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
```

InoSiX, 2015-12-29:
Going to second this error. I am not using RPi however, if that counts for much. same lines are causing the issue (626 and 60 in perform_update).

ayan4m1, 2015-12-30:
Stop OctoPrint first.

```
./venv/bin/pip uninstall OctoPrint-GPX
```

Start it again and install the plugin again. Does that resolve your issue?

markwal, 2015-12-30:
Oops, forgot to attach the file to the release.  New installs will work but upgrades wouldn't.

I think I've got it fixed now.  Testing...

markwal, 2015-12-30:
I think that fixed it.  Please reopen/comment if not.

closed 2015-12-30 by markwal

InoSiX, 2015-12-30:
Interestingly enough the Windows version flags the GPX plugin for update, and the update fails as it does for OP. A fresh install fails with multiple syntax and undeclared identifiers. I am in the (venv) in the octoprint directory when running the command, if that is significant. Likely the update flags the Windows install for the update and it's not compatible?

markwal, 2015-12-30:
@InoSiX, how did you install your Windows version originally?  IE what path and/or from source?

InoSiX, 2015-12-30:
https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows

I must stress I am not skilled with installing things in this manor. So the problem I am having with GPX may be on my end. I may very well be placing it in the wrong place. I see my Octoprint-GPX folder in the root of the Octoprint folder and in the scripts folder of venv.

I am not sure if I should be in (venv) during the install, I am not sure which directory to be in on the command line (or even if it actually matters). 

However, the link above I followed to a tee, and Octoprint runs, and I have the web stream active as well. I believe your credentials will still work.

InoSiX, 2015-12-30:
I also only did the install as you instructed in the previous forum on Google Code. I did nothing else.While I was troubleshooting the failure, I attempted to run the setup.py, it fails with multiple syntax/identifier errors. That might have a lot to do with the python force quitting upon connecting. Only time it fails is as it connects. The interface seems to work. Also, the only thing the GPX log outputs is that log mode was enabled from settings. Suggests to me it isn't getting any further. Is it possible it's the compiler you mentioned previously?

markwal, 2015-12-30:
Ah.  You installed the experimental package I mentioned here: https://groups.google.com/d/msg/octoprint/MQtdDN8QN0s/pYFaMCdoBwAJ I guess?

Shoot.

InoSiX, 2015-12-30:
Yeah, I was that guinea pig ^_^ This is the Windows install.

InoSiX, 2015-12-30:
Just a heads up, I don't think you intended that release for Windows. But with the experimental package installed, Octorprint told me about the new release. Just failed with the errors of OP