markwal/OctoPrint-GPX #12: GPX and Cura Slicing

InoSiX, 2015-12-31:
Freshly installed RPi installation & GPX plugin. What is the proper way to provide a Cura.ini for this device? I assumed the plugin would interact with Cura, hence the reason for the plugin requesting various details relating to the printer. I am unsure how to proceed with my FFCP. Any guidance?

markwal, 2015-12-31:
The best way to get a cura.ini to upload is to install Cura (15.04) on your desktop and use it for a while and to tune in your settings.  Then after it is working well, you can then upload the cura.ini file from your desktop to OctoPrint to do slicing in OctoPrint using those fixed settings.  Any tweaking though and you're back to the desktop.  You can save multiple profiles for the slicing settings that you use most often.

To tune your settings, you slice to gcode on the desktop and then upload that gcode to OctoPrint and print that.  The plugin will translate it to x3g.  To get Cura to work with a MakerBot or clone is a bit of a trick though because you need to get good start/end gcode dialed in.  The reason this is tricky is that Cura emits "RepRap" flavor gcode, so you need the start/end gcode to also be "RepRap" flavor.  On that note, you also need the GPX plugin to be set to "reprap" flavor.

closed 2015-12-31 by markwal

InoSiX, 2015-12-31:
I noticed you happen to use it the same printer. Although it may require additional tweaking, can you post your particular ini?

markwal, 2016-01-02:
Sure.  Probably will need tweaking.  I'm still not happy with it myself and I keep messing around with it whenever I use old Cura.  Not much lately though since I've been trying to get the new Cura to work (and for real printing I still mostly use Slic3r and work around the gap fill bug).