markwal/OctoPrint-GPX #13: Misaligned origin issue when converting gcode with octoprint

chuckmitchell, 2016-01-11:
Any prints I upload to octoprint end up printing to the left of where I see them in the Gcode Visualizer.
I plated the 20mm calibration cube in slic3r and exported gcode, as well as using GPX locally to create an x3g file.

I uploaded the gcode using the octoprint web interface, loaded it and printed. The print was left of center on my bed by about 30mm.

I then saved the .x3g file converted in Slic3r onto the sd card and printed. The box was dead center on my bed.

I've been trying to compare settings in my local copy of gpx with the one included with the plugin, as well as tracing the GCode commands sent to the printer but have been scratching my head.

I'm using a FlashForge creator pro with Sailfish v7.7. 
I do not have any tool offsets, my bed dimensions are as they should be, and are the same as I have them in slic3r on my mac. I don't see an origin input as I do when configuring my printer in slic3r, is it possible this is calculated incorrectly?

chuckmitchell, 2016-01-11:
Here is the output from the octoprint terminal when my print begins.

I'm still getting the hang of GCode, so I might be way off but I see a warning that might be related.

    Send: N12 M132 X Y Z A B*104
    Recv: // echo: (line 11) warning G92 emulation unable to determine all coordinates to set via x3g:140   set extended position
    Recv: // echo: current position defined as
    Recv: // echo: X:212.40 Y:148.04 Z:0.00 A:0.00 B:0.00

Terminal output:

    Send: M70 P15 (Octoprint in controlWe <3 Making Things!)
    Recv:  T0:32 /0 T1:66 /0 B:44 /0
    Recv: ok
    Send: M105
    Recv: ok T0:33 /0 T1:66 /0 B:44 /0
    Send: M105
    Recv: ok T0:33 /0 T1:65 /0 B:44 /0
    Send: M105
    Recv: ok T0:33 /0 T1:64 /0 B:44 /0
    Send: M105
    Recv: ok T0:33 /0 T1:63 /0 B:43 /0
    Changing monitoring state from 'Operational' to 'Printing'
    Send: N0 M110 N0*125
    Recv: ok
    Send: N1 M127*38
    Recv: ok
    Send: N2 M109 S110*106
    Recv: ok
    Send: N3 T1*56
    Recv: ok
    Send: N4 M73 P0*83
    Recv: ok
    Send: N5 M109 S110 T0*41
    Recv: ok
    Send: N6 M104 S230 T1*39
    Recv: ok
    Send: N7 M103*38
    Recv: ok
    Send: N8 G21*18
    Recv: ok
    Send: N9 G162 X Y F6000*68
    Recv:  T0:33 /0 T1:62 /230 B:43 /110
    Recv:  T0:33 /0 T1:62 /230 B:43 /110
    Recv:  T0:33 /0 T1:62 /230 B:47 /110
    Recv:  T0:32 /0 T1:62 /230 B:45 /110
    Recv:  T0:32 /0 T1:63 /230 B:48 /110
    Recv:  T0:33 /0 T1:64 /230 B:48 /110
    Recv: ok
    Send: N10 G161 Z F9000*11
    Recv:  T0:33 /0 T1:67 /230 B:50 /110
    Recv:  T0:33 /0 T1:67 /230 B:50 /110
    Recv: ok
    Send: N11 G92 Z0*104
    Recv: ok
    Send: N12 M132 X Y Z A B*104
    Recv: // echo: (line 11) warning G92 emulation unable to determine all coordinates to set via x3g:140 set extended position
    Recv: // echo: current position defined as
    Recv: // echo: X:212.40 Y:148.04 Z:0.00 A:0.00 B:0.00
    Recv:  T0:33 /0 T1:70 /230 B:50 /110
    Recv: ok
    Send: N13 G90*34
    Recv: ok
    Send: N14 G1 Z30*100
    Recv: ok
    Send: N15 G1 X-95 Y-73 Z30 F14000*63
    Recv: ok
    Send: N16 G130 X0 Y0 A0 B0*30
    Recv: ok
    Send: N17 M6*19
    Recv:  T0:33 /0 T1:70 /230 B:50 /110
    Recv:  T0:33 /0 T1:70 /230 B:50 /110
    Recv:  T0:33 /0 T1:75 /230 B:51 /110
    Recv:  T0:33 /0 T1:78 /230 B:50 /110
    Recv:  T0:32 /0 T1:82 /230 B:51 /110
    Recv:  T0:32 /0 T1:86 /230 B:51 /110
    Recv:  T0:32 /0 T1:89 /230 B:52 /110
    Recv:  T0:33 /0 T1:93 /230 B:52 /110
    Changing monitoring state from 'Printing' to 'Operational'
    Recv:  T0:32 /0 T1:97 /230 B:53 /110
    Recv:  T0:32 /0 T1:102 /0 B:53 /0
    Recv:  T0:33 /0 T1:105 /0 B:53 /0
    Recv:  T0:32 /0 T1:108 /0 B:52 /0
    Recv:  T0:32 /0 T1:109 /0 B:52 /0
    Recv:  T0:33 /0 T1:110 /0 B:51 /0
    Autoscroll showing 365 linesScroll to end | Select all
    Suppress M105 requests/responses

chuckmitchell, 2016-01-11:
Accidentally closed this issue, doh!

chuckmitchell, 2016-01-11:
I'm waiting to testing with default bed parameters.
Please ignore until i can confirm my bed sttings in slic3r are not the problem.

Thanks

chuckmitchell, 2016-01-14:
Closing this as it was the result of some bad start gcode I had input into slic3r.

closed 2016-01-14 by chuckmitchell

sciencedude100, 2017-05-23:
How was the GCode settings fixed?