markwal/OctoPrint-GPX #14: error connecting to a replicator dual

amanda6811, 2016-03-14:
i have a Duinotech l4090 with is a copy of the replicator dual and get an error when i try to connect to my printer.

What do i need to do to get it working

Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: , starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M11035
Recv: Sailfish v7.6
Send: N0 M11035
Recv: echo: gcode to x3g translation by GPX
Recv: SD card ok
Changing monitoring state from 'Connecting' to 'Operational'
Recv: T0:27 /0 T1:26 /0 B:24 /0
Recv: T0:27 /0 T1:26 /0 B:24 /0
Recv: ok
Send: N1 M20*16
Unexpected error while writing to serial port: AttributeError: ''NoneType' object has no attribute 'rfind'' @ comm.py:_doSendWithoutChecksum:1752
Changing monitoring state from 'Operational' to 'Error: AttributeError: ''NoneType' object has no attribute 'rfind'' @ comm.py:_doSendWithoutChecksum:1752'
Connection closed, closing down monitor

markwal, 2016-03-14:
Sorry, I thought I knew where that rfind was, but I don't. It isn't the checksum thing.

Could you upload your octoprint.log somewhere (onedrive, dropbox, etc.)? And post a link here?

That'll probably do, but if it doesn't the next info I'll ask you for is the serial.log and plugin_GPX.log.  Unfortunately those are a bit harder because you have to turn on serial logging and verbose GPX logging in settings (one is under serial communication, the other under GPX).

Also, it looks like you have some Settings->Features set differently than the defaults and than me.  Perhaps it is one of those that reproduce the bug.  In particular, I have "Send Checksums" to be only while printing (And all the Repetier flags are set off).  Where it looks like yours is sending them all the time.  If turning that off fixes the problem, could you post back here so I can track down the bug?

amanda6811, 2016-03-15:
here is a link to my dropbox for the files

https://www.dropbox.com/sh/cs8j15du7f3v5bi/AAALUSp2UITfPV5261Tgw_1ba?dl=0

markwal, 2016-03-21:
Sorry I took so long to look into this.  I was away for a few days.

There are several problems here.  One is that your config.yaml has a lot of settings for Repetier.  The other is that your gpx.ini has some settings that GPX doesn't like.  It may be that there is a bug in the GPX plugin configuration that writes an incorrect gpx.ini.  And if we can reproduce what settings you changed in the UI that created the broken gpx.ini, that'd be good.  If you get a chance, could you grab that file and put it in the dropbox as well?  ~/.octoprint/data/GPX/gpx.ini 

To get you going though, I think you want to start with fresh settings.   Assuming you are running from an octopi image for the Raspberry Pi:
```
ssh pi@youripaddress
sudo service octoprint stop
mv config.yaml configbu.yaml
mv users.yaml usersbu.yaml
mv data/GPX/gpx.ini data/GPX/gpxbu.ini
sudo service octoprint restart
```
The go to the browser and bring up OctoPrint.  Go through the starting wizard to create a new account, etc.  Don't connect yet.  Go to Settings->GPX and choose "Replicator 1 Dual" and click Save.  Now choose the port (/dev/ttyACM0) and 115200 as the baud rate. And click connect.

Now you may want to create a printer profile in Settings->Printer Profiles that has the Origin at the Center, Volume as (225mm, 150mm, 155mm), a heated bed, two extruders and Axis jog speeds (2500mm/min, 2500mm/min, 1100mm/min, 1600mm/min).  Leave the nozzle offsets at 0 since the firmware takes care of it.

markwal, 2016-03-29:
@amanda6811, how's it going?  Did you get it to work?

markwal, 2016-05-06:
Closing due to lack of info.

closed 2016-05-06 by markwal