markwal/OctoPrint-GPX #16: OctoPrint is not aware of jobs finishing when printing from SD Card

david-vo, 2016-03-24:
# What were you doing?
Printing a long job from SD card, initiated through the OctoPrint interface. 

# What did you expect to happen?
I expected OctoPrint to realize the job was done. I checked on it about 2-3 hours after the job was complete and OctoPrint still didn't know it was done.

# What happened instead?
Nothing happened. OctoPrint just acted like the job was still going. I manually typed in the `M73 P100` command in the Terminal tab and sent it. Nothing changed. Later, I checked the end gcode and sure enough it already had that entry at the very top so I've been running it all along.

What's kind of annoying with OctoPrint not realizing that a job ends is the timelapse continues running until you manually intervene.

# Branch & Commit or Version of OctoPrint and GPX Plugin
Version: 1.2.10 (master branch)
GPX 2.4.2

ericleone, 2016-03-31:
I am experiencing the same issue. Seems that 1.2.10 really did a number to this plugin.

markwal, 2016-04-01:
OK, I repro, looks like something has changed in response parsing because the plugin reports Done printing, but OctoPrint doesn't notice.  Should be easy to fix.  Sorry haven't been SD printing as much lately so I didn't notice.

markwal, 2016-04-01:
Should be fixed by ac856d0 as long as I constructed the 2.4.3 release correctly

closed 2016-04-01 by markwal