markwal/OctoPrint-GPX #17: Extruder cold restracts with Simplify 3D G-code

jchristman75, 2016-05-31:
I have a Monoprice Architect that uses the GPX format.

When I start a print it will retract the entire time it runs this command:

G1 X-120 Y-70 Z30 F4800

This code doesn't seem to have an extruder component, and it is before the hotend is heated.

Here is the entire start-build g-code, prior to the problem.  Is there some combination of bad g-code that would cause this to cold retracts and often strip the filament, when I start a print?  Is it possibly a problem with setting offsets?

M73 P0 ; Enable build progress
G130 X127 Y127 Z40 A127 B127 ; Default stepper Vrefs
G162 X Y F3000 ; Home XY maximum
G161 Z F1200 ; Home Z minimum
G92 Z-5 ; Set Z to -5
G1 Z0 ; Move Z to 0
G161 Z F100 ; Home Z slowly
M132 X Y Z A B ; Recall home offsets
M135 T0 ; Load right extruder offsets
**G1 X-120 Y-70 Z30 F4800** ; Move to wait position off table

jchristman75, 2016-05-31:
BTW, I am using the thingverse sailfish.  Are there issues with this firmware?
http://www.thingiverse.com/thing:32084

markwal, 2016-05-31:
Sailfish is fine.  The difficulty is that you need to zero the extruder positions because of the way GPX wants to work with the long running process.

That G92 Z-5 has to set all of the axes because x3g protocol doesn't allow selective axes so GPX makes up the current position.  Offline this would all be zeros.

So change the G92 Z-5 to G92 X0 Y0 Z-5 A0 B0.  Then I also recommend changing the M132 to only recall X Y Z since the A and B offsets are zero anyway and this lets GPX in on the secret:

So to recap, the G92 line becomes:

```
G92 X0 Y0 Z-5 A0 B0
```

And the M132 line becomes:

```
M132 X Y Z
```

closed 2016-05-31 by markwal

jchristman75, 2016-08-21:
Thank you so much! I will try this.  I actually switched to AstroPrint and that works great, but it is so minimal.  I really miss Octo

Do you have any plans to add the ability for GPX to read and use x3g files directly, like AstroPrint does?

markwal, 2016-08-22:
No plans for that actually.

jchristman75, 2016-08-26:
Also, after a print finishes, if I start a new print it will move VERY slowly, and jerky.  Do you know what might cause this?  Do you have a solution for this issue?

If I turn the printer off and turn it back on, it will print correctly, but I don't like power cycling every print.

Thanks.

markwal, 2016-08-26:
Hmmm...  Haven't seen that happen.   The cause might be that the number of ok's that I generated for your end gcode left it out of sync with OctoPrint.   Do you have a small print that you have the gcode for that repros this problem (like a 20mm calibration cube perhaps)?  Also, if you could copy what you see in the terminal window when it is slow and jerky and paste it here that would be helpful.