markwal/OctoPrint-GPX #18: Failed Install

jrpmedia, 2016-06-02:
I'm not a guru so here is the install log for My RPi Model B using OctoPrint: 1.2.10

OctoPrint
		 Settings 
		 System  
		 pi  
 Connection
Serial Port
AUTO/dev/ttyACM0
Baudrate
AUTO2500002304001152005760038400192009600
Printer Profile
Default

Save connection settings

Auto-connect on server startup
Connect
 State
Machine State: Error: Failed to autodetect serial port, please set it manually.
File:  
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: -
Printed: -

 Print
 Pause
 Cancel
 Files

Free: 5.5GB / Total: 7.0GB
 Upload

 Upload to SD

Hint: You can also drag and drop files on this page to upload them.
		Temperature 
		Control 
		GCode Viewer 
		Terminal 
		Timelapse 

0
50
100
150
200
250
300

Actual T: -

Target T: -

Actual Bed: -

Target Bed: -

Actual
Target
Offset
Hotend
off
 °C Set 
 °C Set
Bed
off
 °C Set 
[... 154 lines cut ...]
creating build/lib.linux-armv6l-2.7/octoprint_GPX/static/less
copying octoprint_GPX/static/less/gpx.less -> build/lib.linux-armv6l-2.7/octoprint_GPX/static/less
creating build/lib.linux-armv6l-2.7/octoprint_GPX/templates
copying octoprint_GPX/templates/GPX_settings.jinja2 -> build/lib.linux-armv6l-2.7/octoprint_GPX/templates
copying octoprint_GPX/static/less/Makefile -> build/lib.linux-armv6l-2.7/octoprint_GPX/static/less
running build_ext
building 'gpx' extension
creating build/temp.linux-armv6l-2.7
creating build/temp.linux-armv6l-2.7/GPX
creating build/temp.linux-armv6l-2.7/GPX/src
creating build/temp.linux-armv6l-2.7/GPX/src/pymodule
creating build/temp.linux-armv6l-2.7/GPX/src/shared
creating build/temp.linux-armv6l-2.7/GPX/src/gpx
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c GPX/src/pymodule/gpxmodule.c -o build/temp.linux-armv6l-2.7/GPX/src/pymodule/gpxmodule.o -DGPX_VERSION="\"OctoPrint\"" -DSERIAL_SUPPORT -fvisibility=hidden -IGPX/build/src/shared -IGPX/src/shared -IGPX/src/gpx
arm-linux-gnueabihf-gcc: error: GPX/src/pymodule/gpxmodule.c: No such file or directory
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4

----------------------------------------
Done!
Close

Something went wrong
Installing the plugin from URL "unknown" failed: Could not parse output from pip

markwal, 2016-07-01:
If I recall correctly you solved this via a thread on Google Groups?

DaAwesomeP, 2016-10-16:
> Error: Failed to autodetect serial port, please set it manually.

There's your problem. You need to select the baudrate manually instead of using the `AUTODETECT` option.

closed 2016-10-27 by markwal