markwal/OctoPrint-GPX #20: Second extruder offset

adamgosikwolfe, 2016-07-01:
I've been running a Flashforge Creator Pro with this and I'm having trouble with the second extruder offset. I set the printer properties for the Flashforge dimensions including a 34mm offset for the left extruder. I usually only print on the left extruder and what's been happening is that every print with only the left extruder is 34mm to the left on the buildplate. I've even tried a -34mm offset to no avail. So far I haven't printed anything that this has been an issue with fortunately.  Thoughts on how to fix this? Thank you!

markwal, 2016-07-01:
It is a result of the offline and online GPX being subtley different on how it handles start gcode.  This arises due to an acceleration workaround that only takes effect if the current position is known.  Could you paste your start gcode from your slicer?  I suspect you can fix this by reordering the start gcode slightly.

adamgosikwolfe, 2016-07-01:
Ok sure, here it is. Thanks.

; **** Replicator 1 dual start.gcode ****
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
G1 X150 Y-70 Z30 F9000 ; move to wait position right hand side of the table
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M126 S[fan_speed_pwm]
M140 S[bed0_temperature] T0
M134 T0 ; stabilize bed temperature
M104 S[extruder1_temperature] T1
M133 T1 ; stabilize left extruder temperature
G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
M135 T1 ; left tool
G1 X110 Y-70 E30 F300 ; move nozzle
G1 X120 Y-70 Z0.15 F1200 ; slow wipe
G1 X110 Y-70 Z0.5 F1200 ; lift
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****

markwal, 2016-07-01:
OK, I think it is merely that your wait position is too far right.  When it hits that M133 T1 it will try to move the head 34mm to the right, but the endstop will stop it and then you'll be off by 34mm.  As an experiment, try changing the line G1 X150 Y-70 Z30 F9000 to G1 X112 Y-70 Z30 F9000

markwal, 2016-07-01:
Also, your start gcode has higher than normal stepper vrefs.  The MakerBot and FlashForge typical values are 118 (except Z which is 40).  They'll run hot and noisy at 127 and may melt their mounts.

markwal, 2016-07-01:
Since you've got a FFCP, it has sufficient power in the power supply to heat the bed and the nozzles at the same time without reducing the stepper vrefs, so you could remove the G130 X20 Y20 Z20 A20 B20 line and move the M134 T0 to after you set the extruder temperature, if you wish.

adamgosikwolfe, 2016-07-02:
Thank you, that's a lot of useful information. I always used the default code generated by Simplify3D and didn't think to change the Vrefs.. Will confirm next week when I'm back at work. Thank you!

adamgosikwolfe, 2016-07-12:
That worked perfectly, thanks for the help!

closed 2016-07-12 by adamgosikwolfe