markwal/OctoPrint-GPX #21: Bed Temperature freezes on local screen

sonnylowe, 2016-07-17:
Hello,

 When I start a print, both the extruder and the bed temps will be around 24C. As Marlin increases both temps the extruder temp and bed temps change on my LCD controller. However, only the extruder temps is changing on the temp tab of the browser window. Even though the bed is heating up the bed temp remains at the starting temp (~24C) in Octoprint, this causes the print to stall as Octoprint doesn't think it's at temp and continues the "heating" process. Sometimes I can disconnect and reconnect and the problem goes away, but here lately it is not working anymore...

markwal, 2016-07-17:
You don't need the GPX plugin (this github project is the plugin project), if you're using Marlin.  Did you find the wrong github project?

I suspect you've got a problem with your firmware.  Watch your terminal tab when this happens (or turn on serial logging) and copy and paste that into the OctoPrint support forum: https://groups.google.com/forum/#!forum/octoprint

closed 2016-07-17 by markwal

sonnylowe, 2016-07-18:
Mark,

 My bad, I found this via a google search. I am running Marlin, I just posted this in the wrong spot...thanks for the advice, though, I'll log the problem post the results on the link you provided.

Thanks again,
Sonny