markwal/OctoPrint-GPX #25: Not showing all temps on flashforge style printer

davidknag, 2016-09-09:
Hi, I'm cross posting from octoprint repo https://github.com/foosel/OctoPrint/issues/1485
#### What were you doing?

Printing with my left print head, the right print head unheated. (though it happens always) Looking at the temperature graph.

#### What did you expect to happen?
Show the temperatures for all 3 thermistors (right head, left head, heat bed).

#### What happened instead?
Only the right extruder head, listed as 't0', was shown. I'd like to note that on simplify3d T0 is the number for the heated bed, but I'm not sure how/if they convert over when compiled. The bed was heated but no temp was shown.

#### Branch & Commit or Version of OctoPrint
OctoPrint: 1.2.15 

#### Printer model & used firmware incl. version
Monoprice Architect (Flashforge Creator), upgraded with dual extruders and heated bed. I think it runs an atmega2560.

profile: number of extruders=2, heated bed=true
#### Browser and Version of Browser, Operating System running Browser
Chrome Version 52.0.2743.116 m (64-bit)
#### Link to octoprint.log
unsure how to get a log. Please let me know if you'd like to see.

#### Link to contents of terminal tab or serial.log
I mean.. it's printing. so probably only movement gcode in here
http://pastebin.com/gvcVGsQJ

#### Link to contents of Javascript console in the browser
n/a, console looks fine.

#### Screenshot(s) showing the problem:
https://i.imgur.com/nDKNsvm.png

markwal, 2016-09-09:
You have to edit the printer profile in settings and tell Octoprint you have two extruders.  Also, your eeprom settings need to say there are two. If you think that's all set then what I need is copy and paste from the terminal window when you are *not* printing. In particular, the response from M105.

markwal, 2016-09-09:
The 2 extruder profile must also be set as current (has the star).

BTW I have two extruders and see all temps in the graph. Also, in the code even simplify3d calls the right extruder T0. It must since it uses GPX to translate to x3g.

markwal, 2016-09-09:
Just looked at your graph and I misunderstood your complaint. Your gcode from Simplify3D doesn't match the flavor in GPX settings. If flipping the flavor doesn't help, then what I need to see is your start gcode.

davidknag, 2016-09-09:
yes this fixed it

markwal, 2016-09-09:
Ok, cool.

closed 2016-09-09 by markwal