markwal/OctoPrint-GPX #28: Thing-O-Matic disconnects.

Spackstor, 2016-11-04:
Have a thing-o-matic, pre-sailfish (SD no longer works so had to go back), I get this result:
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: <GPX.gpxprinter.GpxPrinter instance at 0x6fb5b648>, starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110 N0*125
Unexpected error while writing to serial port: IOError: 'Command not supported or recognized' @ comm.py:_do_send_without_checksum:2035
Recv: echo: gcode to x3g translation by GPX
Recv: SD init fail
Changing monitoring state from 'Connecting' to 'Offline: IOError: 'Command not supported or recognized' @ comm.py:_do_send_without_checksum:2035'
Connection closed, closing down monitor
Closing down send loop.

markwal, 2016-11-04:
My GPX plugin is sending an s3g/x3g code that isn't recognized by your firmware.  I'm guessing this may be because your pre-sailfish doesn't support the x3g opcodes that GPX assumes the firmware supports.  GPX works on MakerBot firmware, but maybe not as old as you're running.  We may not be able to fix this without significant investment of development time.

1. What version of the MakerBot firmware are you running?
2. You could turn on verbose GPX logging (Settings->GPX->Verbose) and try again, then grab the GPX plugin log from Settings->Logs, upload it somewhere and put a link here.

I don't want to get your hopes up though.  The result of looking at the logs is likely to be: yep, GPX sends opcode x and it'd be difficult to avoid that and test all the cases (I don't have a TOM), so I can't help you.

Spackstor, 2020-01-21:
Got the thingomatic working. It has to run slowly so I don't use it much. Have an ORD rova, a much better machine.

closed 2020-01-21 by Spackstor