markwal/OctoPrint-GPX #30: Printing issue and cancel print issue

Brianp82, 2016-12-01:
Hey so I have a Qidi Tech and my friend has a Flashforge. We both use this plugin so we can connect to our printers via Octoprint and upload gcode to print. The first problem is when printing through OctoPrint the printer starts to deviate from the start script and extrude tons of filament and do huge retracts. Its not the same everytime. Happens on both of our printers. And if we print from the SD card this never happens. Also, when we cancel a print through OctoPrint, the communication with the printer is lost. I get no error, but I cannot control my bed or start another print or anything unless I recycle power on either the Octo or printer. Any help would be appreciated. Thanks.

markwal, 2016-12-17:
A couple of separate problems here.

First, the retracts.  That happens because OctoPrint is one session with GPX and a problem with the protocol.  The problem with the Sailfish/Makerbot protocol is that there is no way to set only some axes with G92.  So many people's start gcode accidentally sets the extruder position.  Compounding this problem is that GPX always wants to do relative extruder moves so computes the relative motion based on what it thinks the current position is, which can be different than the slicer in this configuration.  There's an easy fix by modifying your start gcode slightly.  I can help you do that with two pieces of info: What slicer are you using?  And what is your start gcode?

The second problem of needing to click the disconnect button (and connect after that) after a cancel is a known race condition that I haven't been able to figure out yet.  The underlying problem is that either the host or the bot can initiate a cancel and the bot responds the same way in either case so to avoid an infinite cancel loop, I have to have a state that is "waiting for the bot to acknowledge cancel", but sometimes I miss that ack and the protocol for this is very fragile since it only sends it once.  The way I had it working originally is to always disconnect on cancel and then the user has to click connect.  I was thinking of putting it back that way and have an option that turns on this experimental "stay connected after cancel" situation.

Brianp82, 2016-12-18:
Hey thanks for the reply. My slicer program is S3D and my start script is this(default)
M73 P0 ; Enable build progress
G162 X Y F3000 ; Home XY maximum
G161 Z F1200 ; Home Z minimum
G92 Z-5 ; Set Z to -5
G1 Z0 ; Move Z to 0
G161 Z F100 ; Home Z slowly
M132 X Y Z A B ; Recall home offsets
M135 T0 ; Load right extruder offsets
G1 X-110 Y-70 Z30 F9000 ; Move to wait position off table
G130 X20 Y20 Z20 A20 B20 ; Lower stepper Vrefs while heating
M126 S[fan_speed_pwm] ; Set fan speed
M140 S[bed0_temperature] T0 ; Heat buildplate 
M134 T0 ; Stabilize bed temperature
M104 S[extruder0_temperature] T0 ; Heat right extruder
M104 S[extruder1_temperature] T1 ; Heat left extruder
M133 T0 ; Stabilize right extruder temperature
M133 T1 ; Stabilize left extruder temperature
G130 X127 Y127 Z40 A127 B127 ; Default stepper Vrefs
G92 A0 B0 ; Zero extruders
G1 X100 Y-70 F9000 ; Move to front right corner of bed
G1 Z0.3 F6000 ; Move down to purge
G1 X-90 Y-70 E24 F2000 ; Extrude a line of filament across the front edge of the bed
G1 X-100 Y-70 F180 ; Wait for ooze
G1 X-110 Y-70 F5000 ; Fast wipe
G1 Z0.5 F100 ; Lift
G92 A0 B0 ; Zero extruders
M135 T1 ; Load left extruder offsets
G1 X-100 Y-68 F9000 ; Move to front left corner of bed
G1 Z0.3 F6000 ; Move down to purge
G1 X90 Y-68 E24 F2000 ; Extrude a line of filament across the front edge of the bed
G1 X100 Y-68 F180 ; Wait for ooze
G1 X110 Y-68 F5000 ; Fast wipe
G1 Z1 F100 ; Lift
G92 A0 B0 ; Zero extruders
M73 P1 ; @body (notify GPX body has started)

Also, in the GPX plugin settings in Octoprint, should I select replicator 1 dual extruder or Flashforge? I have a Qidi-Tech which is technically a clone of both. Not sure if that even makes a difference. What settings should be selected in the settings screen anyway?

markwal, 2016-12-18:
Yes. If you change two lines, it should fix it.

Change the first G92 Z-5 to
G92 X0 Y0 Z-5 A0 B0

And then drop a and b off of the M132 line so it is
M132 X Y Z

That keeps the extruders always in the known state.  That should fix it.

markwal, 2017-06-28:
Hopefully 2.6.0 fixes the issues to be fixed here (and I'm assuming the start gcode solved the other).

closed 2017-06-28 by markwal