markwal/OctoPrint-GPX #32: Gcode temps disregarded

pearson222, 2016-12-15:
Loading gcode that has been sliced in Makerbot Desktop.  Temperatures set for both tools and bed in slicer are disregarded.  Slicer set for ABS (225C T0 and 115C Bed) upon starting print T0 begins preheat for 115C and Bed for 0.  

Refer to the issue logged @ Octoprint Github. 

OctoPrint#1655
https://github.com/foosel/OctoPrint/issues/1655

markwal, 2016-12-15:
BTW, is your gcode flavor set to "MakerBot" now or did my plugin reset it to "RepRap" on upgrade?

pearson222, 2016-12-15:
It is set to Makerbot.  I just rebooted Octoprint, power cycled the Pi and printer and it seems to be working now...

pearson222, 2016-12-15:
Not sure what was happening.  You can go ahead and close the ticket and I will reach back out if it starts acting up again.

markwal, 2016-12-15:
I know what causes this behavior, I'm just not sure what happened to get into this state.  The underlying difficulty is that MakerBot and RepRap disagree over what M109 means.  A RepRap slicer (and RepRap firmware understands) puts out M109 to mean: set the nozzle temperature and wait for it to get there.  MakerBot uses it to mean, set the bed temperature.  So Makerbot Desktop puts out something like:

M104 S225
M109 S115

And if GPX is in RepRap mode, it sees: set the nozzle to 230 degrees and don't wait.  Now set it to 110 and wait for it.  And nothing happens to the bed at all.

The thing we'd want to track down here is, how does the setting get reverted back to RepRap flavor if you've been using MakerBot.  Weird.

One idea I've thought about is when the gcode is setting a nozzle temperature to something that looks like a bed temperature, maybe I should pop up a notification and offer to switch the flavor on the fly.  Or maybe just detect the flavor from the comments in the gcode, or something.  Haven't got around to it though.

closed 2016-12-15 by markwal