markwal/OctoPrint-GPX #33: Machine Type Creation Request

topdarkstar, 2016-12-30:
I wanted to thank you for creating this plugin to make life using OctoPrint that much easier.
I noticed that you have the FlashForge Creator Pro model already added to GPX as a supported machine type, and would like to request if you could add the FlashForge Dreamer as well.

The primary differences between these models that would need to be added to a separate machine type is:
I acquired the specifications from the FlashForge Dreamer user guide.
[Document](http://www.flashforge.com.hk/uploads/5/8/1/9/58196085/dreamer_user_guide_en_v3.0_05.01.2015.pdf)
[Website Specs](http://www.flashforge.com/dreamer-3d-printer/tech-specs/)

Print Volume:
230 mm (X)
150 mm (Y)
140 mm (Z)

Maximum Speed for the Axes:
2000 mm/min (X)
2000 mm/min (Y)
200   mm/min (Z)
300   mm/min (E)

Nozzle Diameter:
0.4 mm

2 Extruders
Nozzle Offset
T1=
34 mm (X)
0   mm (Y)

markwal, 2016-12-31:
My understanding is the plugin doesn't work at all with a Dreamer.  It doesn't use the same serial protocol.  I don't know for sure, since I don't have one.  Do you have it working except for the info you give here?  Or are you asking for it to work at all with the Dreamer?

topdarkstar, 2017-01-06:
Greetings, Mark. I have received my printer and have been able to use it the last three days over usb and with the included wifi module using the provided FlashPrint Software. I would like to see what I can do to help you get this printer working if possible.
I had thought that this printer used the same protocol for USB/Serial as the Creator Pro, but here's what I have so far I can give you. I can also provide some of the Profile information as well.
Baud Rate 115200 bits/sec

setExtruderTempGcode,M104 S$ T#
stabilizeExtruderTempGcode,M109 S$ T#
setBedTempGcode,M140 S$
stabilizeBedTempGcode,M190 S$
reportTempGcode,M105
fanPowerGcode,M106 S$
fanOffGcode,M107
movementOverrideGcode,M220 S$
extrusionOverrideGcode,M221 S$
laserOnGcode,M600 S255
laserOffGcode,M601
syncLaserGcode,M400
initializeSD,M21
listSDfiles,M20
selectSDfile,M23 $
startSDprint,M24
pauseSDprint,M25
reportSDstatus,M27
homeAxisGcode,G28
toolchangeGcode,M108 T$
emergencyStopGcode,M112
enableMotorsGcode,M17
disableMotorsGcode,M18
waitForStartupCommand,1
startupCommand,start
serialAcknowledge,ok
commentChar,;
dwellGcode,G4 P$
macro1name,Macro 1
macro1,G91 ; relative positioning,G1 Z50 ; lift 5 cm,G90 ; absolute positioning
macro2name,Macro 2
macro2,
macro3name,Macro 3
macro3,
allowCommandBuffering,1
serialCacheSize,63
includeLineNumbersAndChecksums,0
flowControlMethod,0
useCommunicationTimeout,1
timeoutMilliseconds,30000
GPXconfig,r2
XstepsPerMM,88.573186
YstepsPerMM,88.573186
ZstepsPerMM,400.000000
AstepsPerMM,96.2752018703
BstepsPerMM,96.2752018703
AaxisHasHBP,0
BaxisHasHBP,0

topdarkstar, 2017-01-06:
Atmel JTAG ICE mkII
and
AVR Dragon (avr910-programmer)
Default Baudrate is 19200, but after the latest firmware update; it changes (without notification to 115200)

I believe it even supports bitbanging on the serial port.

It defaults Parallel to lpt1 and it defaults serial to com1

topdarkstar, 2017-01-06:
Here's all the Configuration Settings from the embedded slicer (ffslicer):
# ffslicer default is config file

precision = 0.1                 # valid range [0.01, 0.4], default is 0.1
filamentDiameter = 1.75         # valid range [0.5, 10.0], default is 1.75
extrusionRatio = 1.09           # valid range [0.5, 2.0], default is 1.0
layerHeight = 0.2               # valid range [0.01, 0.4], default is 0.18
firstLayerHeight = 0.27         # valid range [0.05, 0.5], default is 0.27
pathWidth = 0.4                 # valid range [0.35, 0.45], default is 0.4
baseSpeed = 80                  # valid range [10, 200], default is 80 
travelSpeed = 120               # valid range [10, 200], default is 120 
minSpeed = 5                    # valid range [1, 20], default is 5 
rightTemp = 200                 # valid range [0, 260], default is 200
leftTemp = 200                  # valid range [0, 260], default is 200
platTemp = 50                   # valid range [0, 120], default is 50
vaseMode = false				# valid value is {true, false}, default is false 
lineNumber = false				# valid value is {true, false}, default is false 

[dimension]
enable = false                  # valid range {true, false}, default is false
innerOfs = 0.1                  # valid range [-1.0, 1.0], default is 0.1
outerOfs = 0.0                  # valid range [-1.0, 1.0], default is 0.0

[shell]
count = 2                       # valid range [0, 10], default is 2
outerMaxSpeed = 40              # valid range [10, 80], default is 40
outerSpeedRatio = 0.5           # valid range [0.1, 1.0], default is 0.5
visibleInnerSpeedRatio = 0.7    # valid range [0.1, 1.0], default is 0.7
invisibleInnerSpeedRatio = 1.0  # valid range [0.1, 1.0], default is 1.0
innerOverlapRatio = 0.3         # valid range [0.0, 0.5], default is 0.3
seamSpeedRatio = 0.15           # valid range [0.05, 0.3], default is 0.15
decelerateDist =  10.0          # valid range [0.0, 20.0], default is 10.0
randStart = false               # valid value is {true, false}, default is false 
startClosestX = -10000.0        # valid range [-10000.0, 10000.0], default is -10000.0
startClosestY = -100.0          # valid range [-10000.0, 10000.0], default is 0.0

[infill]
topSolidCnt = 3                 # valid range [1, 30], default is 3
bottomSolidCnt = 3              # valid range [1, 30], default is 3
density = 0.15                  # valid range [0.0, 1.0], default is 0.15
shape = hexagon                 # valid valule is {line, triangle, hexagon}, deafult is hexagon
overlapRatio = 0.15             # valid range (-0.7, 0.7), default is 0.15
startAngle = 45                 # valid range [0, 180], default is 45
topMaxSpeed = 70                # valid range [10, 200], default is 70 
solidSpeedRatio = 0.7           # valid range [0.1, 1.0], default is 0.7 
sparseSpeedRatio = 1.0          # valid range [0.1, 1.0], default is 1.0 
combineCnt = 2                  # valid range [1, 4], default is 2
combineSparseOnly = true        # valid value is {true, false}, default is true 

[raft]
enable = false                  # valid range {true, false}, default is false 
extruder = 0                    # valid range {0, 1}, default is 0
margin = 5.0                    # valid range [3.0, 10.0], default is 5.0
liftHeight = 0.25               # valid range [0.1, 0.4], default is 0.3
bottomHeight = 0.4              # valid range [0.2, 0.4], default is 0.4 
bottomWidth = 2.0               # valid range [0.4, 4.0], default is 2.0 
bottomDensity = 0.5             # valid range [0.05, 1.0], default is 0.5 
bottomSpeed = 10                # valid range [10, 30], default is 10 
middleHeight = 0.2              # valid range [0.1, 0.4], default is 0.2
middleWidth = 0.4               # valid range [0.1, 0.4], default is 0.4
middleDensity = 0.3             # valid range [0.05, 1.0], default is 0.3
middleSpeed = 60                # valid range [30, 80], default is 60 
topHeight = 0.2                 # valid range [0.2, 0.4], default is 0.2 
topWidth = 0.4                  # valid range [0.1, 0.4], default is 0.4 
topDensity = 1.0                # valid range [0.2, 1.0], default is 1.0 
topSpeed = 80                   # valid range [30, 120], default is 80
bottomTempAdjust = 5            # valid range [0, 20], default is 5

[preExtrude]
enable = true                   # valid range {true, false}, default is true 
[... 50 lines cut ...]
voltageZ = 40                   #  valid range [20, 100], default is 40 
voltageA = 100                  #  valid range [60, 120], default is 100 
voltageB = 100                  #  valid range [60, 120], default is 100 

[svg]
baseTime = 3.0                  # valid range [1.0, 60.0], default is 3.0 
attachLayer = 4                 # valid range [0, 10], default is 4 
attachTime = 5.0                # valid range [1.0, 60.0], default is 5.0 
lightIntensity = 1.0            # valid range [0.01, 2.0], default is 1.0 
adjustX = 100.0                 # valid range [50.0, 150.0], default is 100.0 
adjustY = 100.0                 # valid range [50.0, 150.0], default is 100.0 

[svgRaft]
enable = true                   # valid range {true, false}, default is true
margin = 10.0                   # valid range [5.0, 15.0], default is 10.0
layerCnt = 6                    # valid range [3, 15], default is 6

[svgInfill]
enable = false                  # valid range {true, false}, default is false
topSolidCnt = 40                # valid range [10, 300], default is 40
bottomSolidCnt = 40             # valid range [10, 300], default is 40
shellThickness = 2.0            # valid range [1.0, 10.0], default is 2.0
infillThickness = 1.0           # valid range [0.5, 5.0], default is 1.0
density = 0.2                   # valid range [0.0, 1.0], default is 0.15
zPrecision = 0.1				# valid range [0.025, 0.25], default is 0.1

markwal, 2017-01-06:
I don't think the GPX plugin does anything useful for a Dreamer.  The GPX plugin translates from ASCII gcode to the MakerBot x3g protocol.  As far as I can tell, the Dreamer consumes gcode directly.  So what is needed is a comm plugin that can send the ASCII gcode directly to a Dreamer through whatever communication protocol you select (perhaps wifi or ethernet for example).  This would use very little of the existing code, if any.  It's really a new project.  It might be fun if I had a Dreamer, but I don't so there isn't really a way for me to help since nothing about the Creator Pro relates.

closed 2017-01-06 by markwal