markwal/OctoPrint-GPX #34: Octoprint fails to print on center of bed

InMuSights, 2017-01-04:
Have a Monoprice Flashforge Creator with default firmware. Using Chrome. OP is Version 1.3.0 (master branch).

I have attempted to print from OP but the offset ?? is putting the center to the back right corner of the bed.

If I use Makerbot software to print the items, i have tried many different stls', is centered but attempts to use Cura and Slic3er upload gcode is always offset unless I radically shift the image in Cura Desktop.

Next I thought that if I exported a profile to OP that it might print but the same thing happens. In CuraEngine 15.04 there is a check to use Origin 0,0 and selecting that appears to center but the bed travels to the bottom of the build area - thus wasting material.

Can someone help getting this working with Octoprint?

My starting gcode ....
; -- START GCODE --
M136 ; start build
M73 P0
G90 ; absolute coordinates
;
; set temperatures and assert Vref
M140 S{print_bed_temperature}
M104 S{print_temperature} T0
G130 X118 Y118 A118 B118 ; set stepper motor Vref to defaults
; let the Z stepper vref stay at eeprom level (probably 40)
;
; home and recall eeprom home position
T0 ; home on the right nozzle
G28 X Y Z ; home all axes at homing speed
G92 X0 Y0 Z0 A0 B0 ; set all coords to 0 for now
G1 Z5 F500 ; move Z 5mm away so we can carefully hit the limit switch
G161 Z F100 ; home Z slowly
M132 X Y Z ; recall stored home offsets for XYZ axes
;
; wait for heat up
G1 X110 Y-72 Z30 F3300 ; move to waiting position
M116 ; wait for temps
;
; purge and wipe
G92 E0 ; set current extruder position as 0 so that E15 below makes sense
G1 X110 Y-70 Z0.2 F2400.0 ; move to just on the bed
G1 X110 Y70 E15 F1200.000 ; extrude a line of filament along the right edge of the bed
G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this 0
;
; Sliced at: {day} {date} {time}
; Basic settings: Layer height: {layer_height} Walls: {wall_thickness} Fill: {fill_density}
; Print time: {print_time}
; Filament used: {filament_amount}m {filament_weight}g
; Filament cost: {filament_cost}
; -- end of START GCODE --

No ideas on next step...

markwal, 2017-01-06:
A few things need to be true:  

**Slicer Origin**
One is that you have to tell the slicer that the origin (0,0) is at the center of the bed.

**Home Offsets**
Because your start gcode uses M132 to set the coordinate system, your home offsets (distance from the center of the bed to the home position) need to be stored in EEPROM (the LCD has menu items for this)  X should be approx 150+mm and Y should be approx 75+mm if I recall correctly

**Visualization**
By default, OctoPrint in the gcode viewer uses the RepRap convention of having the origin on the lower left hand corner.  If you don't tell it that the origin is in the center in printer profiles then it'll show the print on the lower left hand corner of the bed, but if you print it, it'll show up in the right place.

This isn't a bug, but a configuration problem.  I'm happy to help, so go ahead and continue the questions here, but I'm going to close the issue since there's no "to do" for me to fix (but you can still comment).

closed 2017-01-06 by markwal

InMuSights, 2017-01-12:
Just looked at my printer and there is no way to set any home offsets in the LCD eeprom.

Not sure what cfg is wrong.

markwal, 2017-01-12:
http://www.sailfishfirmware.com/doc/parameters-home-offsets.html

InMuSights, 2017-01-12:
I am not using Salfish - its a Monorice firmware.  I am not seeing an option to change the offsets.

markwal, 2017-01-12:
It's a fork of Sailfish. They may have hidden it from the LCD, but hopefully you can access it using other software as in the manual.  You can also just override it in the gcode by using G92 with the correct offsets rather than M132.

markwal, 2017-01-12:
The way you can tell it is a fork is that GPX works at all. No sailfish, then GPX can't connect.

InMuSights, 2017-01-12:
so you are saying that in my GCODE were i have M132 X Y Z ; recall stored home offsets for XYZ axes I should have something like G92 X150 Y75 Z0 A0 B0 ; ?

I tried that and it acts the same

InMuSights, 2017-01-12:
I would love to upgrade my Monoprice Architect to use latest Salfish but I have found no references online to anyone doing that.....

markwal, 2017-01-13:
It's your slicer configuration then not the offsets.  I took this picture of Cura 15.04 Machine Settings:

[image]

markwal, 2017-01-13:
You could also upload a gcode file somewhere (dropbox, google drive, onedrive) and share a link here.  I could take a look and tell you definitively.

InMuSights, 2017-01-15:
Mark it looks like it is working now.
Buddy bought a Qidi and his gcode worked on my machine.
Thanks for your help...

johnryanlee, 2017-09-25:
I cannot get my Geeetech Prusa I3 (GT2560 board) to have +20 X-axis offset?

Where would this be set?  Must I use GCODES?

markwal, 2017-09-25:
@johnryanlee Geetech Prusa I3 uses a Marlin variant firmware, does it not?  I don't think you need GPX for that printer.

johnryanlee, 2017-09-26:
@markwal yes I think it does. I am asking on the wrong thread

I loaded up marlin 1.1.5 last night and the issue remains that it doesn't print from center.  It always starts in front left and i need a 20mm offset so that it prints on the bed and not in the air.  Did lots of reading last night and it appears I need to set the X_MIN_POS to -20 because my x min endstop switch is there. 

Does that sound right?  While I have your attention, i see marlin heats up the bed and hotend simultaneously which the original firmware would do it separately. Slight concern that it did that to save the load on the components?  I do see that it takes longer to heat up the bed which implies it can't feed a much current directly because it's solution the power.

markwal, 2017-09-26:
The X_MIN_POS thing sounds plausible to me.

The heat up is a matter of your start g-code.  You can have your start g-code heat them separately if that's better for your electronics by changing the start g-code in the slicer.