markwal/OctoPrint-GPX #35: FFCP: Steps Per MM calibration

totalitarian, 2017-01-24:
My 20mm test cubes come out at X20.05mm by Y19.9mm so I think I need to adjust the Y Axis steps per mm. Do I do this in my firmware, slicer (Simplify3d which produces gcode) or the the machine settings for this plugin?

I did try to increase my Y steps per mm by 2mm in the GPX plugin as a test but the cube came out the same size.

markwal, 2017-01-28:
The 20mm test cubes are really more useful for figuring out whether you are under or over extruding based on the concavity or convexity of the top.  It is also useful for figuring out some other print faults.  It's really too small for figuring out if you have a steps/mm problem since you may just have an edge problem (ie outer most extrusion is protruding too far due to over extrusion, etc.) instead of a scale problem.

To check scale, you want to print a couple of layers of a large rectangle (hollow should be fine).  There are some examples on thingiverse.

For a printer that uses the x3g protocol, the thing that defines your steps per mm, is the tool you are using to translate from gcode to x3g.  If you are printing x3g off of the SD card using this plugin, the adjustment needs to be made in the tool that generated that x3g (RepG, MakerBot Desktop, or GPX.ini on your desktop).  If you are printing gcode directly from OctoPrint, then it is this plugin's settings that matter under "machine settings".

If you increase your Y steps per mm by 2, it should shrink the distance that the carriage travels in Y by a little less than 2%.  However, you may need to disconnect and reconnect (button in "Connection" box) after changing that setting for it to take effect.

totalitarian, 2017-01-28:
Thanks, I actually have this issue on larger size prints too. I found this formula to work out the correct steps to use.

(Desired size*CurrentStepsPerMMValue)/(Measured Size)

I've rebooted both boxes several times and changing the steps per /mm seems to have no effect at all :(

markwal, 2017-01-29:
Please share the following:

1. What output from Simplify3D do you use?  A gcode file or an x3g file?
2. Do you print from gcode on the pi or x3g on your machine's SD card?
3. Does it retain your machine settings when you reboot?

totalitarian, 2017-01-29:
1) I use gcode and and run in through gpx.exe with a custom gpx.ini to define the steps per mm (see below)

2) Both - when I use gcode on the pi - the steps/mm are ignored, when I use the SD card with the x3g generated by gpx.exe and the custom.ini the steps/mm are correctly applied

3) Yes

```
;************ MACHINE ************
[machine]
; specifies the nominal filament diameter (either 1.75 or 3.0)
nominal_filament_diameter=1.75
; specifies the number of extruders on this machine
extruder_count=2
; sets the timeout for homing in seconds
timeout=20

;************ X AXIS ************
[x]
; sets the maximum feedrate for this axis in mm/s
max_feedrate=18000
; sets the home feedrate for this axis in mm/s
home_feedrate=2500
; sets the number of steps per mm of movement for this axis
; Pulley dia: 10.82mm / 1/8 step = 1/(10.82 * pi / 1600)
steps_per_mm=94.139704
; sets the homing direction for this axis
; maximum = 1
; minimum = 0
endstop=1

;************ Y AXIS ************
[y]
; sets the maximum feedrate for this axis in mm/s
max_feedrate=18000
; sets the home feedrate for this axis in mm/s
home_feedrate=2500
; sets the number of steps per mm of movement for this axis
; Pulley dia: 10.82mm / 1/8 step = 1/(10.82 * pi / 1600)
steps_per_mm=95.090610
; sets the homing direction for this axis
; maximum = 1
; minimum = 0
endstop=1

;************ Z AXIS ************
[z]
; sets the maximum feedrate for this axis in mm/s
max_feedrate=100000
; sets the home feedrate for this axis in mm/s
home_feedrate=1600
; sets the number of steps per mm of movement for this axis
; TR-8x8 Z axis = 1/(8/1600)
steps_per_mm=400
; sets the homing direction for this axis
; maximum = 0
; minimum = 1
endstop=1

;************ RIGHT EXTUDER (A AXIS) ************
[a]
; sets the maximum feedrate for this axis in mm/s
max_feedrate=1600
; sets the number of steps per mm of extrusion
; Steps/mm is calculated by dividing the 'drive gear steps per revolution'
; (in this case, equal to motor_steps) by the 'drive gear circumference'
; (drive gear diameter = 10.14) So we get: 1600/(PI * 0.14) = 50.235....
steps_per_mm=96.275201870333662468889989185642
; sets the number of steps per revolution
motor_steps=1600
; signals if this tool has a heated build platform
has_heated_build_platform=1

;************ LEFT EXTUDER (B AXIS) ************
[b]
; sets the maximum feedrate for this axis in mm/s
max_feedrate=1600
; sets the number of steps per mm of extrusion
steps_per_mm=96.275201870333662468889989185642
; sets the number of steps per revolution
motor_steps=1600
; signals if this tool has a heated build platform
has_heated_build_platform=0
```

markwal, 2017-01-30:
Sorry.  You indeed have discovered a bug.  I've reproduced, the gpx uses the wrong ini path for the machine definition in the current version.

totalitarian, 2017-01-30:
No problem. Hope it's not too difficult for you to fix.

markwal, 2017-01-30:
OK.  In my case, the bug turns out to be that when I choose "FlashForge Creator Pro" as the machine name, under the covers gpx uses that as a synonym for "Replicator 1 Dual" and it tries to load "r1d.ini", but the machine editor saved it as "ffcp.ini".  I wrote the synonym stuff and the machine editor, but I didn't make them match.

Perhaps this isn't the problem you are seeing though.  What machine did you choose in the GPX settings UI?

totalitarian, 2017-01-30:
It is interesting you mentioned that. Until about an hour ago I was using the FFCP profile, I then switched to Replicator Dual, changed the steps and my test piece came out perfect. I was about to post mentioning this but it seems you beat me to it!

markwal, 2017-02-01:
This bug will be fixed in the next release