markwal/OctoPrint-GPX #38: Print after successful print causes extreme slow movement

fform, 2017-02-02:
Printer: Makerbot Rep 2X
Firm: Sailfish v7.7
Slicer: Simplify 3D 3.1.1
Printing gcode from Raspberry pi local via Octoprint 1.3.1, GPX 2.5.5

I can print successfully with Octoprint and the GPX plugin. 

Normally, a print will move the carriage to the back right corner, bring the bed up quickly, back down a bit and then slowly touch bring it up to touch the nozzle. Then it quickly moves the carriage to the front left corner, drops the bed a bit and starts to heat up.

After a good print via Octoprint's local storage & the GPX plugin, the next print I try to start will move the carriage to the back right and bring the platform up like normal, but then the carriage move to the front left will be really slow. Disconnecting at this point will not stop the print. Rebooting octoprint and reconnecting spits out a log like this
```
Recv:  T0:21 /0 T1:22 /0 B:43 /110
Recv:  T0:21 /0 T1:22 /0 B:44 /110
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: <GPX.gpxprinter.GpxPrinter instance at 0x71fe8620>, starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110 N0*125
Recv:  Sailfish v7.7
Send: N0 M110 N0*125
Recv: echo: gcode to x3g translation by GPX
Recv: SD init fail
Recv:  T0:21 /0 T1:22 /0 B:45 /110
Recv:  T0:21 /0 T1:22 /0 B:46 /110
```
The printer is just heating up like nothing happened but Octoprint doesn't know anything about it with the status stuck on "Connecting"

The only thing that will actually stop this zombie printing is by rebooting the PI. Once I do that, I can start hte print again, no problem and get this log
```
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: <GPX.gpxprinter.GpxPrinter instance at 0x71fe8620>, starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110 N0*125
Recv:  Sailfish v7.7
Recv: echo: gcode to x3g translation by GPX
Recv: SD init fail
Recv:  T0:22 /0 T1:23 /0 B:106 /110
Send: N0 M110 N0*125
Recv:  T0:22 /0 T1:23 /0 B:106 /110
...
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok
Recv: // echo: (line 1) Syntax warning: unsupported mcode command 'M115'
Send: N2 M21*18
Recv: ok
Recv: SD init fail
Send: N3 M105*36
Recv: ok T0:23 /0 T1:23 /0 B:108 /110
Send: N4 M105*35
Recv: ok T0:23 /0 T1:23 /0 B:108 /110
Send: N5 M105*34
Recv: ok T0:23 /0 T1:23 /0 B:108 /110
Send: N6 M105*33
Recv: ok T0:23 /0 T1:23 /0 B:109 /110
Send: N7 M105*32
Recv: ok T0:23 /0 T1:23 /0 B:109 /110
Send: N8 M105*47
Recv: ok T0:23 /0 T1:23 /0 B:109 /110
Send: N9 M105*46
Recv: ok T0:23 /0 T1:23 /0 B:109 /110
Send: N10 M105*22
Recv: ok T0:23 /0 T1:23 /0 B:109 /110
Send: N11 M105*23
Recv: ok T0:23 /0 T1:23 /0 B:110 /110
Send: N12 M105*20
Recv: ok T0:23 /0 T1:23 /0 B:110 /110
Send: N13 M105*21
Recv: ok T0:23 /0 T1:23 /0 B:110 /110
Send: N14 M105*18
Recv: ok T0:23 /0 T1:23 /0 B:110 /110
Changing monitoring state from 'Operational' to 'Printing'
Send: N0 M110 N0*125
Recv: ok
Send: N1 G90*17
Recv: ok
Send: N2 M83*26
Recv: ok
Send: N3 M73 P0*84
Recv: ok
Send: N4 G162 X Y F3000*76
Recv:  T0:23 /0 T1:23 /0 B:110 /110
Recv:  T0:23 /0 T1:23 /0 B:110 /110
```

Let me know if this makes sense, or if you need additional logs. Thanks!

markwal, 2017-02-02:
I know about the can't cancel a heatup from OctoPrint (it's a result of a change to OctoPrint last November conflicting with how the plugin does cancel).  Also, connecting to a bot that is doing something doesn't stop it from doing that thing (like preheating).  That's a feature.  :-)

The bug here is the slow move.  What I'd like to see is what the last few lines of the successful print look like.  Also, are you using the left nozzle for anything and does it make a difference if you do?

If you could attach the gcode file of the successful first print, that'd be helpful (upload to DropBox or similar and share a link here).

ramonimbao, 2017-02-13:
Adding `G92 A0 B0` to the "Before print job starts" gcode scripts under settings seems to have fixed this for me.

markwal, 2017-02-13:
@ramonimbao That's a good work-around actually.  Ensures that GPX and OctoPrint are all in agreement about the position of the extruders at the beginning of a print.  I use it on my non-GPX printers as well since messing around with the extrusion controls in OctoPrint changes the absolute position of them.

Thanks for adding that here for others to find.

closed 2017-02-13 by markwal

fform, 2017-02-16:
I don't think that's right, or maybe it's just in the wrong spot for my gcode. I have that in a gcode file.Here's a sample gcode file I used and the process
https://gist.github.com/fform/908a7c448a307f5a81ec4952650f8531

1. Generate above gcode with Simplify 3d.
2. Upload to octoprint, local, PI, storage. Printer turned on, connected to Octoprint, load gcode and print.
3. Printer moves to back right, does the bed move and then quickly moves to the front left where everything starts to heat up.
4. Prints successfully.
5. Press print again in octoprint, same gcode this time, but same behavior with any gcode file, previously successful or not (all generated from Simplify 3d)
6. Extruder moves to back riht, bed comes up, drops a bit and slowly comes up (like normal)
7. Extruder then tries to move further right (and back?) and clicks, then very slowly starts moving to the front left. Octoprint reports temp on the bed is increasing while the extruder is still moving.
This is the starting log from that https://gist.github.com/fform/3aac5d4f62084c5ff788f1f94513784f

The only way to get it out of that state is to power cycle the printer.
8. Power cycle, load the print and push print. Works as described in step 3-4. This is the starting log from that https://gist.github.com/fform/bfe07dab3a14b13925823090b29a5b06

I see that `G92 A0 B0` in the gcode a few times before printing begins so I'm not sure that's the problem I'm seeing. Let me know if you need more details