markwal/OctoPrint-GPX #39: Octoprint z-changes

Jabarnes01, 2017-02-08:
We are seeing a problem with our replicator 2 where the z-change event is not firing. Sometimes it does but only after the 4th z-change. Does anyone else experience this issue?

markwal, 2017-02-08:
What event do you mean?  Did you write a plugin that is listening for an event?  Or do you mean the command line's tied to events via config.yaml?  Or do you mean that the timelapse pictures aren't being taken?

There is a setting that specifies the minimum Z change required to do some of those things.  That setting is to prevent Z-lift (slicer feature) from triggering a layer change event.  Perhaps your layer height is 1/4 of the minimum Z change?

markwal, 2017-04-22:
I'm closing this for now, but feel free to add more info and I'll reopen.

closed 2017-04-22 by markwal