markwal/OctoPrint-GPX #40: Can’t add a new machine definition

ospalh, 2017-02-14:
Looking at the GPX (w/o OctoPrint ) Readme, you can use `-c` to use a new machine definition.
I have found just a file like that for my printer, a Makerbot clone from CTC, and can’t tell the plugin to use it, but have to modify an existing file, which is error-prone.

markwal, 2017-02-26:
True.  The plugin doesn't have a feature for uploading an .ini file.

Changing the title to reflect a feature request.

markwal, 2017-02-26:
BTW, I believe the CTC is a clone of a Replicator 1 Dual, so you can also just choose that from the drop down.

ospalh, 2017-02-26:
The point is that it’s a Replicator 1 clone, with *different* stepper and extruder values.
I selected the Replicator 1 and had to change the values. Works, but could be simpler.