markwal/OctoPrint-GPX #41: Can't install

orc886, 2017-02-24:
I can't find GPX on the plugin manager so I copied and pasted the link and got this error:

"Building wheels for collected packages: OctoPrint-GPX
Running setup.py bdist_wheel for OctoPrint-GPX: started
Failed building wheel for OctoPrint-GPX"

I'm running on windows right now. Can it be the reason?

markwal, 2017-02-26:
Yes, currently the plugin is only supported on Linux.  It's possible to get it to build on Windows by installing various tools, but I believe the last time somebody tried it crashed.  We'd need a developer/contributor to iterate on the Windows build to get it to work and then volunteer to test and fix any future releases since I don't have a Sailfish printer connected to a Windows machine to test on.

orc886, 2017-02-26:
That makes sense now. I might buy a Pi and test out.

Also, I'm not sure if that's a GPX issue, but when I try to connect through OctoPrint I get weird question mark symbols from my machine.
Is that something the GPX plugin solves?

markwal, 2017-02-26:
If you have a Sailfish based 3D printer (FlashForge Creator [not Dreamer], CTC Dual, Powerspec Pro [not Ultra], MakerBot Replicator 2 [not Gen 5 or later], etc.), you need the GPX plugin for OctoPrint since Sailfish talks a binary protocol (x3g) which looks like garbage (question marks, diamonds, blocks, etc.) in a text terminal.

closed 2017-02-26 by markwal

orc886, 2017-02-26:
Makerbot Replicator 2

markwal, 2017-02-26:
Yes, OctoPrint requires the GPX plugin to talk to the Makerbot Replicator 2.   Actually that's the default printer setting for GPX because that was the printer the original author (@whpthomas) had when he created GPX.

orc886, 2017-02-26:
Thanks a bunch!

jordan-gleeson, 2020-03-24:
Hi Mark, has there been any progress since 2017 on windows compatability? I have a requirement where a Windows machine is my only option.

markwal, 2020-04-26:
@jordan-gleeson sorry, not really.