markwal/OctoPrint-GPX #43: 2nd Print Ini Failing

RJ-Make, 2017-03-05:
This is a strange problem so hopefully I can articulated it well.

On a first print after a machine and OctoPrint reboot the print head and bed preposition; and on heat up the print head is driven forward as normal.  

On a second print the process is the same, however this time the print head moves forward very slowly and the extruder is turned on. This of course eats the filament causing a no fee issue when the print starts. 

When this cycle happens, it appears a reboot is the only "fix"

Let me know what logs you need.

markwal, 2017-03-06:
You don't need to do a reboot.  Disconnect and reconnect work.  This is a side-effect of x3g's G92 behavior interacting with your start.gcode.  The first time through, the X Y Z A B coordinates are all zero as far as GPX knows.  On the second time, there are values left over and unfortunately they get sent to your printer because in x3g G92 always takes all coordinates, so when your start gcode says G92 Z0, in reprap land that means just zero the Z coordinate and leave the other ones alone.  GPX simulates this by sending what it thinks all the other coordinates are which are whatever are leftover.  

If you share your start gcode here, I can help you craft a workaround.

markwal, 2017-03-06:
You might also find this thread helpful: https://groups.google.com/d/msg/octoprint/8Gws5Rv-ueQ/2n9ZZMFTBAAJ

RJ-Make, 2017-03-07:
Thanks Mark, here's the S3D starting script. I'll also take a look at the url you provided.  I could have sworn I tried disconnecting and reconnecting, but will make it a point to try that again.

Again thanks for a great plug-in and help!

`; **** Replicator 1 dual start.gcode ****
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
M132 X Y Z A B ; recall home offsets
M135 T0 ; load right extruder offsets
G1 X150 Y-70 Z30 F9000 ; move to wait position right hand side of the table
G130 X20 Y20 Z20 A20 B20 ; lower stepper Vrefs while heating
M126 S[fan_speed_pwm]
M140 S[bed0_temperature] T0
M134 T0 ; stabilize bed temperature
M104 S[extruder0_temperature] T0
M133 T0 ; stabilize right extruder temperature
G130 X127 Y127 Z40 A127 B127 ; default stepper Vrefs
G92 A0 B0 ; zero extruders
G1 Z0.4 ; position nozzle
G1 X110 Y-70 E25 F300 ; purge nozzle
G1 X120 Y-70 Z0.15 F1200 ; slow wipe
G1 X110 Y-70 Z0.5 F1200 ; lift
G92 A0 B0 ; zero extruders
M73 P1 ;@body (notify GPX body has started)
; **** end of start.gcode ****`

markwal, 2017-03-07:
I think the main thing is to fully specify the G92's and avoid recalling extruder positions since that just confuses GPX. So...

Change G92 Z-5 to G92 X0 Y0 Z-5 A0 B0
And M132 X Y Z A B to M132 X Y Z

The other G92's on the extruders should be ok because they follow a G1 with all coordinates so those will be what is sent for the G92.

You could also make other improvements like your vrefs are too high (127 instead of 118) and you don't need to lower them for heat up. And you can heat the bed and the nozzle at the same time, but I think that thread covers those issues.

closed 2017-03-07 by markwal

RJ-Make, 2017-03-07:
Thanks for all your help, one last question. Will this work for dual extrusion?

> ; **** Replicator 1 dual start.gcode Left ****
T0; always home on the right extruder
M73 P0 ; Enable build progress
G162 X Y F3000 ; Home XY maximum
G161 Z F1200 ; Home Z minimum
G92 X0 Y0 Z-5 A0 B0 ; Set Z to -5
G1 Z0 ; Move Z to 0
G161 Z F100 ; Home Z slowly
M132 X Y Z ; Recall home offsets
G1 X-95 Y-70 Z30 F9000 ; Move to wait position off table
T1 ; switch to the left extruder
M126 S[fan_speed_pwm] ; Set fan speed
M140 S[bed0_temperature] T0 ; Heat buildplate
M104 S[extruder0_temperature] T0 ; Right
M104 S[extruder1_temperature] T1 ; Left
M116 ; Wait for all temps to be reached
G130 X118 Y118 Z40 A118 B118 ; Default stepper Vrefs
G92 A0 B0 ; Zero extruders
G1 X100 Y-70 F9000 ; Move to front right corner of bed
G1 Z0.3 F6000 ; Move down to purge
G1 X110 Y-70 E25 F300 ; purge nozzle
M135 T1; left tool
G1 X120 Y-70 E25 F300; purge nozzle
M135 T0; right tool
G1 X120 Y-70 Z0.15 F1200 ; slow wipe
G1 X110 Y-70 Z0.5 F1200 ; lift
G92 A0 B0
M73 P1 ; @body (notify GPX body has started)
; **** end of start.gcode ****

markwal, 2017-03-08:
I think so I didn't examine all the purges coordinates closely, so as long as that does the bead you want where you want, it looks ok to me.

At the moment, @body will be ignored by OctoPrint which means that it's hard to get some macros to work properly (@pause at z will get confused by the moves in the start gcode for example).  I've got a todo on my list to make something for that.

As a note, the PWM value can't be translated to x3g.  The x3g protocol can only turn the fan on or off, so PWM > 0 means ON and PWM=0 means OFF and that's it.  Sailfish 7.8 Beta does have a fan PWM setting in EEPROM, but it just defines what ON means.

RJ-Make, 2017-03-14:
Thank you Mark!  You have done an amazing job, not only the work in the plug-in but in the support you provide!