markwal/OctoPrint-GPX #46: Can only control printer while octoprint print job is running

adcurtin, 2017-05-05:
I have a powerspec 3d pro (flashforge creator pro clone) with sailfish on it.

When I connect with GPX, the printer starts a 'print'. I am then able to preheat the printer just fine. To start my print on the printer, I need to cancel the octoprint 'print'. However, once I do this, octoprint can no longer control my printer (even after my manually started print is finished). Octoprint displays the temperature correctly though, so it is still getting data from it.
I have to disconnect and reconnect to be able to control it again.

Is this a bug, or working as designed?  If it's the latter, would it be possible to add a convenient reconnect button, or is that not something a plugin could do?

markwal, 2017-05-07:
It's an interaction behavior that's newish with recent versions of OctoPrint.  I used to key off of the existence of line numbers to tell which commands are arriving during an OctoPrint print and which are while OctoPrint is monitoring idle, but now the plugin has OctoPrint send line numbers on everything so it always thinks a print is going.  I'll come up with a fix eventually.

However, you really shouldn't initiate the print from the LCD if you want to monitor from OctoPrint.  To print from the SD card, choose the SD card file from the OctoPrint UI and tell it to start the print from there.  Let me know if you have problems with that since I usually just print from USB these days.

adcurtin, 2017-05-07:
It's been a while since I've tried to do that. I like to be at my printer to start prints and don't usually have a computer convenient. I think last time I tried that, I was having some issues getting the file listing to update (I think I had to disconnect and reconnect) when I changed SD cards. Octoprint also never seemed to display progress for me. But it has been a while, so I'll give it another shot and report back (with better details if I run into issues).

adcurtin, 2017-05-09:
I couldn't get octoprint to refresh the file listing after changing the SD card. I had to disconnect and reconnect. When printing, I had no indication of progress (which I think is expected with x3g).

markwal, 2017-05-09:
Huh. The refresh button above the file list didn't work? Weird.

Yeah, no progress. Dan and I have talked about sneaking a byte offset into an x3g protocol packet to get progress but haven't got around to it yet.