markwal/OctoPrint-GPX #47: Unable to connect via USB

synfinatic, 2017-07-11:
This seems to be a re-occuring problem with my FlashForge Creator X (configured as Replicator 1 Dual extruder).  Even after rebooting my Pi & Printer, trying to connect to USB gives me at least a 50% chance of connection failure.  The error in the terminal is:

Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'
Unexpected error while connecting to serial port: AUTO IOError: 'AUTO port and baudrate not currently supported by GPX' @ comm.py:_openSerial:1754 (hook GPX)
Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'

In setup/Serial Connection, I have set the baud to 115200 and serial port /dev/ttyACM0, but for some reason in the Connection dialog on the main page the serial port is changed to "AUTO".  Baud rate stays at 115200.

No idea why, but the problem seems to have gotten noticeably worse since I added a relay to control the power of the printer.  I don't think it has anything to do with the issue, but I'm including it for completeness.  

Once I successfully connect, it seems stable and I'm able to print off SD without any problems.

markwal, 2017-07-12:
You may find the PortLister plugin helpful.  The problem you are having is if the printer isn't all the way on and connected to the pi at the precise moment you visit or refresh the web page, the port won't be in the list and when you press connect, the web page will send "AUTO" to OctoPrint.  You won't get a successful connect until you can select the proper port from the box and you won't be able to do that until you refresh the page *after* the printer is all the way on with a solid USB connection.

The PortLister plugin could help because it will try to keep that dropdown up to date with the state of the pi.

If that doesn't solve it for you, then I'd suggest a different USB cable on a different path from the pi to the printer.

synfinatic, 2017-07-13:
Wow.  That totally worked.  Thanks!

closed 2017-07-13 by synfinatic