markwal/OctoPrint-GPX #48: Can't install from Octoprint

laurentRueil, 2017-08-26:
Hello,
I'm thinking to buy a Raspberry PI 3 to manage and see the extruder dring the print of my Flashforge Creator Pro.
Before buying it, I want to try it on a HP tablet ElitePad with Win10 32B.
All seems to work (I'm in vacation without the 3D printer), I can simulate a print and record a movie.
I can also install some plugins.

But I need GPX for my Flashforge.
I have an issue during the installation of the plugin.

Here the log file (reduced)

2017-08-26 12:34:19,331 C:\OctoPrint\venv\Scripts\python.exe -m pip install https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
2017-08-26 12:34:23,357 > Collecting https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
2017-08-26 12:34:24,723 > Downloading https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz (125kB)
2017-08-26 12:34:33,934 > Requirement already satisfied: OctoPrint in c:\octoprint\venv\lib\site-packages\octoprint-1.3.4-py2.7.egg (from OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,934 > Requirement already satisfied: flask<0.11,>=0.9 in c:\octoprint\venv\lib\site-packages\flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,934 > Requirement already satisfied: Jinja2<2.9,>=2.8 in c:\octoprint\venv\lib\site-packages\jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,936 > Requirement already satisfied: werkzeug<0.9,>=0.8.3 in c:\octoprint\venv\lib\site-packages\werkzeug-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,937 > Requirement already satisfied: tornado==4.0.2 in c:\octoprint\venv\lib\site-packages\tornado-4.0.2-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,937 > Requirement already satisfied: sockjs-tornado<1.1,>=1.0.2 in c:\octoprint\venv\lib\site-packages\sockjs_tornado-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,938 > Requirement already satisfied: PyYAML<3.11,>=3.10 in c:\octoprint\venv\lib\site-packages\pyyaml-3.10-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,940 > Requirement already satisfied: Flask-Login<0.3,>=0.2.2 in c:\octoprint\venv\lib\site-packages\flask_login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,940 > Requirement already satisfied: Flask-Principal<0.4,>=0.3.5 in c:\octoprint\venv\lib\site-packages\flask_principal-0.3.5-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,941 > Requirement already satisfied: Flask-Babel<0.10,>=0.9 in c:\octoprint\venv\lib\site-packages\flask_babel-0.9-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,943 > Requirement already satisfied: Flask-Assets<0.11,>=0.10 in c:\octoprint\venv\lib\site-packages\flask_assets-0.10-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,944 > Requirement already satisfied: markdown<2.7,>=2.6.4 in c:\octoprint\venv\lib\site-packages\markdown-2.6.9-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,944 > Requirement already satisfied: pyserial<2.8,>=2.7 in c:\octoprint\venv\lib\site-packages\pyserial-2.7-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,946 > Requirement already satisfied: netaddr<0.8,>=0.7.17 in c:\octoprint\venv\lib\site-packages\netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,947 > Requirement already satisfied: watchdog<0.9,>=0.8.3 in c:\octoprint\venv\lib\site-packages\watchdog-0.8.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,947 > Requirement already satisfied: sarge<0.2,>=0.1.4 in c:\octoprint\venv\lib\site-packages\sarge-0.1.4-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,948 > Requirement already satisfied: netifaces<0.11,>=0.10 in c:\octoprint\venv\lib\site-packages\netifaces-0.10.6-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,950 > Requirement already satisfied: pylru<1.1,>=1.0.9 in c:\octoprint\venv\lib\site-packages\pylru-1.0.9-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,950 > Requirement already satisfied: rsa<3.3,>=3.2 in c:\octoprint\venv\lib\site-packages\rsa-3.2.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,951 > Requirement already satisfied: pkginfo<1.3,>=1.2.1 in c:\octoprint\venv\lib\site-packages\pkginfo-1.2.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,953 > Requirement already satisfied: requests<2.8,>=2.7 in c:\octoprint\venv\lib\site-packages\requests-2.7.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,953 > Requirement already satisfied: semantic_version<2.5,>=2.4.2 in c:\octoprint\venv\lib\site-packages\semantic_version-2.4.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,954 > Requirement already satisfied: psutil<3.3,>=3.2.1 in c:\octoprint\venv\lib\site-packages\psutil-3.2.2-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,956 > Requirement already satisfied: Click<6.3,>=6.2 in c:\octoprint\venv\lib\site-packages\click-6.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,957 > Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in c:\octoprint\venv\lib\site-packages\awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,957 > Requirement already satisfied: feedparser<5.3,>=5.2.1 in c:\octoprint\venv\lib\site-packages\feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,959 > Requirement already satisfied: chainmap<1.1,>=1.0.2 in c:\octoprint\venv\lib\site-packages\chainmap-1.0.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,960 > Requirement already satisfied: future<0.16,>=0.15 in c:\octoprint\venv\lib\site-packages\future-0.15.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,960 > Requirement already satisfied: scandir<1.4,>=1.3 in c:\octoprint\venv\lib\site-packages\scandir-1.3-py2.7-win32.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,961 > Requirement already satisfied: websocket-client<0.41,>=0.40 in c:\octoprint\venv\lib\site-packages\websocket_client-0.40.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,963 > Requirement already satisfied: python-dateutil<2.7,>=2.6 in c:\octoprint\venv\lib\site-packages\python_dateutil-2.6.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,963 > Requirement already satisfied: itsdangerous>=0.21 in c:\octoprint\venv\lib\site-packages\itsdangerous-0.24-py2.7.egg (from flask<0.11,>=0.9->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,964 > Requirement already satisfied: MarkupSafe in c:\octoprint\venv\lib\site-packages\markupsafe-1.0-py2.7-win32.egg (from Jinja2<2.9,>=2.8->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,966 > Requirement already satisfied: certifi in c:\octoprint\venv\lib\site-packages\certifi-2017.7.27.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,967 > Requirement already satisfied: backports.ssl_match_hostname in c:\octoprint\venv\lib\site-packages\backports.ssl_match_hostname-3.5.0.1-py2.7.egg (from tornado==4.0.2->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,967 > Requirement already satisfied: blinker in c:\octoprint\venv\lib\site-packages\blinker-1.4-py2.7.egg (from Flask-Principal<0.4,>=0.3.5->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,969 > Requirement already satisfied: Babel>=1.0 in c:\octoprint\venv\lib\site-packages\babel-2.5.0-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,970 > Requirement already satisfied: speaklater>=1.2 in c:\octoprint\venv\lib\site-packages\speaklater-1.3-py2.7.egg (from Flask-Babel<0.10,>=0.9->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,970 > Requirement already satisfied: webassets>=0.10 in c:\octoprint\venv\lib\site-packages\webassets-0.12.1-py2.7.egg (from Flask-Assets<0.11,>=0.10->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,971 > Requirement already satisfied: argh>=0.24.1 in c:\octoprint\venv\lib\site-packages\argh-0.26.2-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->OctoPrint-GPX==2.6.0)
2017-08-26 12:34:33,973 > Requirement already satisfied: pathtools>=0.1.1 in c:\octoprint\venv\lib\site-packages\pathtools-0.1.2-py2.7.egg (from watchdog<0.9,>=0.8.3->OctoPrint->OctoPrint-GPX==2.6.0)
[... cut ...]
2017-08-26 12:34:49,134 ! Command "C:\OctoPrint\venv\Scripts\python.exe -u -c "import setuptools, tokenize;file='c:\users\laurent\appdata\local\temp\pip-x8jfpu-build\setup.py';f=getattr(tokenize, 'open', open)(file);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, file, 'exec'))" install --record c:\users\laurent\appdata\local\temp\pip-ii7wix-record\install-record.txt --single-version-externally-managed --compile --install-headers C:\OctoPrint\venv\include\site\python2.7\OctoPrint-GPX" failed with error code 1 in c:\users\laurent\appdata\local\temp\pip-x8jfpu-build\

laurentRueil, 2017-08-27:
Does this install error occur because I'm on win 10 and it's only compatible with Linux OS?
Is it possible to install GPX for windows and have the possibility to use it?

markwal, 2017-10-15:
GPX plugin has only been tested on Linux (and in particular, for me anyway, only under Raspbian on a Raspberry Pi).

closed 2017-10-15 by markwal