markwal/OctoPrint-GPX #50: Octoprint-GPX not remembering setting, thus not printing correctly

Nico19951995, 2017-09-23:
I recently updated to the newest octoprint (0.14) and newest GPX plugin (2.6.0) The plugin now does not remember it's settings(machine type needs to be r1p, g-code flavor markerbot), meaning it would not print correctly. It heats up the wrong printhead, and when printing starts it won't extrude. printer is a FlashForge Creator X. I also tried to change the settings manually in de GPX.ini file, this wouldn't work either. Only thing which works is removing the GPX.ini all together. this fixes the problem for one single print.
Installing a older version of the GPX plugin doesn't work either.

I have read a couple more of these issues here, but couldn't find a proper answer in them.

markwal, 2017-10-15:
I'm guessing that the octoprint user doesn't have permission to write to ini location.  (you meant r1d, right?).  Perhaps you have a broken r1d.ini left around from before?  Perhaps there is something in octoprint.log (an exception or error from GPX)?

edacraneo, 2018-01-07:
hello,
I have the same problem, can't save settings... can you us please ?
than you a lot