markwal/OctoPrint-GPX #51: Bed control always off

GyulaMasa, 2017-10-03:
Hi,

I have a CTC bizer 3D printer with dual heads.
I am running OctoPi on aRaspberry Pi 3 B.
My current problem is, that the heated bed temperature always switches back to "off", what ever I do.
The print starts, however due to the cold bed, the PLA does not stick to the surface, and the head picks the first layer up, to a blob.
The printer otherwise works when attached to my PC.
I installed the GPX plugin, I installed Cura (V15.04.6) onto my PC.
I have set up the configuration file for Cura (using PLA, nozzle temp 220C, bed temp 70C).
I exported the file, as written, and imported it to Octopi.
I also tried to print GCode file, instead of an .stl, but the bed temperature still stays off.
(I checked the Gcode in a text editor, but it looks like the bed temp is not set there)
Usually, I try first with a Google search, but I did not really find any relevant information so far.
Thank you very much!

markwal, 2017-10-15:
To figure this out, I think I'll need a serial log (it's off by default, you turn it on in Octoprint settings).  Or at least copy and paste the relevant bit from the terminal window.

GyulaMasa, 2017-10-15:
Hi Mark,

Please find the part of serial log below!
There were some error messages inside, exactly, where the platform should have been heated up.
I took the liberty to remove most of the report, and also to re-edit the file in order to not to be a heap of characters.

[serial.log.short.formed.txt](https://github.com/markwal/OctoPrint-GPX/files/1385846/serial.log.short.formed.txt)

(I removed the initial heat-up phase and after the error messages, the actual prrint reports.)
However, I recognised, that the command, I added to the "begin script" section had a typo (M140 70 T0, instead the M140 S70 T0)
I fixed it.
However, the same error message is visible again.
[2nd.serial.log.short.formed.txt](https://github.com/markwal/OctoPrint-GPX/files/1385857/2nd.serial.log.short.formed.txt)
It seems, that for some reason thepltfrm considered non-existent.
I hope, this will help you to find out!

GyulaMasa, 2017-10-17:
OK, I tried one more thing:
I generated a gcode file from Makerbot desktop, since I can control the platform from that SW.
This is the beginning of the gcode, generated by MakerBot Desktop:
M136 (enable build progress)
M73 P0
G162 X Y F2000(home XY axes maximum)
G161 Z F900(home Z axis minimum)
G92 X0 Y0 Z-5 A0 B0 (set Z to -5)
G1 Z0.0 F900(move Z to '0')
G161 Z F100(home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
G92 X152 Y75 Z0 A0 B0
G1 X-112 Y-73 Z150 F3300.0 (move to waiting position)
G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
M109 S70 T0
M134 T0
M135 T0
M104 S245 T0
M133 T0
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
; Makerbot Industries
; Miracle Grue 3.9.4
; This file contains digital fabrication directives
; in G-Code format for your 3D printer
; http://www.makerbot.com/support/makerware/documentation/slicer/
; 
; Right Toolhead Weight (grams): 8.27297 
; Right Toolhead Distance (mm): 2711.46 
; Duration: 3644.19 seconds 
; Active extruders in print: 0 
; Chunk 0 
; Lower Position  0 
; Upper Position  0.1 
; Thickness       0.1 
; Width           0.4 

I recognised the following:
The command for setting the bed temperature is M109 S70 T0, instead of M140 S70 T0.
I changed the "startcode" section in Cura accordingly, exported the .ini file, and imported to Octopi.
I started again a print, but is seems, it has the same failure:
(part from the serial log:)
2017-10-17 21:31:39,511 - Send: N6 M109 T0 S250.0*51
2017-10-17 21:31:39,513 - Recv: ok
2017-10-17 21:31:39,514 - Recv: // echo: (line 6) Semantic warning: M109 cannot select non-existant heated build platform
2017-10-17 21:31:39,515 - Send: N7 M109 T1 S250.0*51
2017-10-17 21:31:39,516 - Recv: ok
2017-10-17 21:31:39,520 - Recv: // echo: (line 7) Semantic warning: T1 cannot select non-existant extruder
2017-10-17 21:31:39,521 - Send: N8 T0*50
2017-10-17 21:31:39,521 - Recv: // echo: (line 7) Semantic warning: M109 cannot select non-existant heated build platform
2017-10-17 21:31:39,522 - Recv: ok
2017-10-17 21:31:39,524 - Send: N9 M190 S70.0*72
2017-10-17 21:31:39,526 - Recv: ok
2017-10-17 21:31:39,527 - Recv: // echo: (line 9) Semantic warning: M190 cannot select non-existant heated build platform
2017-10-17 21:31:39,528 - Send: N10 M136 (enable build)*96

This is my Cura "start" code:
G92 X0 Y0 Z-5 A0 B0 (set Z to -5)
G1 Z0.0 F{travel_speed}(move Z to '0')
G161 Z F100(home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
G92 X152 Y72 Z0 A0 B0
G1 X-141 Y-74 Z40 F{travel_speed} (move to waiting position)
G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
M109 S70 T0; Set bed temperature o 70C
M134 T0 ; Wait bed temp to stabilize
M135 T0
M104 S250 T0 ; Set nozzle temp to 250C
M133 T0
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)

This is the generated Gcode file beginning:
;Generated from xyzCalibration_cube.stl d316cc8422b4622151bc32fe2910f6a8731736bb
;Generated with Cura_SteamEngine 15.04.6
M140 S70.0
M104 T1 S250.0
M109 T0 S250.0
M109 T1 S250.0
T0
M190 S70.0

M136 (enable build)
M73 P0
G162 X Y F2000(home XY axes maximum)
G161 Z F900(home Z axis minimum)
G92 X0 Y0 Z-5 A0 B0 (set Z to -5)
G1 Z0.0 F7200(move Z to '0')
G161 Z F100(home Z axis minimum)
M132 X Y Z A B (Recall stored home offsets for XYZAB axis)
G92 X152 Y72 Z0 A0 B0
G1 X-141 Y-74 Z40 F7200 (move to waiting position)
G130 X20 Y20 A20 B20 (Lower stepper Vrefs while heating)
M109 S70 T0; Set bed temperature o 70C
M134 T0 ; Wait bed temp to stabilize
M135 T0
M104 S250 T0 ; Set nozzle temp to 250C
M133 T0
G130 X127 Y127 A127 B127 (Set Stepper motor Vref to defaults)
; Sliced ?filename? at: Tue 17-10-2017 21:31:31
; Basic settings: Layer height: 0.1 Walls: 2 Fill: 20
; Print time: ?print_time?
; Filament used: ?filament_amount?m ?filament_weight?g
; Filament cost: ?filament_cost?
M73 P0;
; -- end of START GCODE --

;Layer count: 198
;LAYER:0
M107
G0 F7200 X11.400 Y11.400 Z0.300
;TYPE:SKIRT

markwal, 2017-10-18:
Ah. You have "Replicator 2" set as your printer type in GPX settings perhaps?  You should set it to "Replicator 1 Dual"
.

A rep2 doesn't have a heated platform and only one nozzle.

Also, if you are using Cura, you should have the flavor set to Reprap and don't use Makerbot desktops start gcode.

GyulaMasa, 2017-10-18:
Hi Mark,

My GPX setting is:
Machine type: Replicator 1 - dual extruder
Gcode Flavor: RepRap
Recalc build progress is YES
Ditto priinting is NO
Filament value is 1.75mm for both
and no extrusion recalculation
When I print, I upload .stl files, and slice them by Octoprint with the
Cura imported .ini files.
In cure, there are two RepRap settings:
- Marlin/Sprinter
- Volumetric.
Now I set it up to the first, and try to print the test cube again.

Thank you for the help!

Regards,

Gyula

markwal, 2017-10-18:
Volumetric won't work very well unless you also fake some volume to linear translation by setting nominal and actual to do the multiplication for you.

GyulaMasa, 2017-10-18:
OK, Then I will not try!
Now it seems to be working.
However, now both extruders are heated up to 250C...
I will figure out, hogy to disable it.

Thanky anyways!
Gyula

GyulaMasa, 2017-10-18:
HI Mark,

Now I have I/O error, and the print stops unexpectedly...

Unexpected error while writing to serial port: IOError: '[Errno 5]
Input/output error' @ comm.py:_do_send_without_checksum:2414Changing
monitoring state from 'Printing' to 'Offline: IOError: '[Errno 5]
Input/output error' @
comm.py:_do_send_without_checksum:2414'Connection closed, closing down
monitor

I updated Octoprint today to the latest version.
However I cannot tell, what causes this issue...

Gyula

2017-10-18 19:50 GMT+02:00 Gyula Masa <gyula.masa@gmail.com>:

mralph03, 2018-07-06:
I'm having a similar problem. I've run my OctoPi for years now, and I didn't change any settings. One day I printed a model with no problems, and the following day the print failed with a line 14 semantics error, claiming there was no heated print bed for the M140 command.

I checked for updates, moved from the stable build to the maintenance build, checked the build plate function directly on the machine (heats fine), and printed with the build plate disabled in my slicer (prints fine through OctoPi).

What has changed that would suddenly cause the OctoPi to stop recognizing my build plate?

markwal, 2018-07-06:
Typically it is because the printer type setting has been reset to Replicator 2.  This can happen if the gpx.ini gets corrupted or deleted.  There was an instance a few months ago that this happened due to an underlying change in one of the packages that octoprint relies on, but you may have tripped over a different case

mralph03, 2018-07-06:
I did check that setting, and it's always been on Replicator 1 (Dual) as long as I've used it. Incidentally, I didn't notice last I set it that there is a FF Creator Pro option, which is what I have. I used this machine definition and it appears to recognize the machine bed.

So my problem is gone, if not solved.