markwal/OctoPrint-GPX #54: Plugin GPX not saving settings

edacraneo, 2018-01-08:
Hello.
Thank you for your excellent job !

But i have a problem : i loose save settings for each reload of octoprint.

In the plugin settings, I configure this one like this: "replicator 2 with HBP" and "Makerbot", and i save. When i reboot the raspberry, the setting is different, as originally, it disappeared.

I tried different ways to record but impossible, my setting disappears each reboot.

Version of OctoPrint

1.3.6 Stable
GPX 2.6.0

Any idéo please ?

Thank you a lot

markwal, 2018-01-08:
I can't reproduce this issue.  Are you installing onto a raspberry pi configured with the octopi image?  If so, which version of that image did you use?  I just tried a fresh install of 0.14 and I was able to save settings.

As an aside, do you actually have a MakerBot Replicator 2 and added your own HBP?

If you have a clone of some kind, even if they say they cloned a Rep 2, they actually cloned the Replicator 1 Dual.  If you use Rep2 with HBP, it'll not match in a few respects, but particularly your prints will be the wrong size.

edacraneo, 2018-01-08:
Hello,
Thank you for this quickly reply, it's really nice.

I have a real makerbot Replicator 2and I added a HBP, so it's a true replicator 2 with same dimensions plate. it should work without any problems right?

I will redo the installation from the beginning...

However, I do not understand why every time I connect to octoprint, my GPX configuration resets itself.

Thank you for your job and help

edacraneo, 2018-01-08:
I reinstalled octoprint, made the updates, and I still have the same problem, however i notice that this bug occurs when i turn off the printer before turning off octoprint. If I turn off my printer first, and then turn off octoprint when I restart Octoprint, my settings are not saved.

I use relays and PI GPIOs to turn off the printer with my phone or pc at work.

If the problem comes from there, would you be able to provide me a file so that by default my plugin is on: "Replicator 2 with HBP" and "Makerbot" please?

Thank you for you precious help

Regards,

markwal, 2018-01-09:
I attempted to replicate with your settings and couldn't make it lose the settings.  When you click "save" on the settings dialog, it should save the settings in question to ~/.octoprint/data/GPX/gpx.ini. You can check via ssh after clicking Save.

Those persist on the microSD card on the Raspberry Pi.  If the persistence isn't working due to your microSD card caching without writing, you may be able to resolve by setting the settings, clicking Save in the settings dialog and then use "reboot the system" from the system menu.  That will go through the normal OS power down procedure so should flush any caches.

If it isn't persisting the settings, the settings don't work at all since the GPX library reads the settings from there (there isn't a separate, in memory only version of them that GPX knows how to use).

markwal, 2018-01-09:
This probably is unrelated, but just in case: the gpx.ini persistence only regards the "printer type" in the GPX settings.  The printer profile chosen on connection is OctoPrint's own setting.  The default for that is set in "Printer Profiles".

edacraneo, 2018-01-10:
Thank you for your help,
I checked gpx.ini after saving, it's OK but sometimes, when i start octoprint, the GPX.ini resets.
I found a temporary solution, I remove the write rights on the GPX.ini file after configuration, octoprint can't change et my setting stay on replicator 2 HBP and makerbot....

aragocom, 2018-02-04:
I am also having problems with the settings not sticking. I had custom steps per mm set, logged out, closed the web page and when I logged back in the x and y had reverted to the 88.88 settings. 
I have previously used GpxUi and it had no problems with the custom settings. 
I have a CTC dual so I am using Clone R1 Dual with HBP.
On the Octoprint Pi the gpx.ini looks OK but the cr1d.ini does not match the cr1d.ini from GpxUI. 

GpxUi -
[machine]
[x]
steps_per_mm=93.460000
home_feedrate=2502
[y]
steps_per_mm=93.300000
[z]
steps_per_mm=402.848400
[a]
steps_per_mm=96.0000000000000000000000000000000
[b]
steps_per_mm=103.8397999999999967712938087061048

On the Pi -
[a]
steps_per_mm=96
[b]
steps_per_mm=101.052699999
[machine]
[x]
[y]
[z]
steps_per_mm=402.848400

I verified that the time on the cr1d.ini matched the time I saved the settings in Octoprint. So it appears the cr1d.ini is not being written correctly ?
Regards

aragocom, 2018-02-04:
Further to the above ...
Printed a test print for accuracy in the x and y axis at 20, 50 and 100mm. With the settings saved from preferences - very poor results. Generally between 2.4 and 5.8% undersize. I did not restart Octoprint after saving the settings.
I then copied the cr1d.ini from the GpxUi folder, made it read only and restarted Octoprint. The test print came out perfect - 0.1mm or less variation at 20 and 50 mm, 1mm or less at 100 mm. 
I can live with that. 
Regards