markwal/OctoPrint-GPX #55: Steps per mm "Y-axis" no change

JKJameson, 2018-02-17:
I have a Flashforge Creator Pro but seeing approx 5% loss in x and y measurements when using gcode instead of x3g.

I have fixed the x-axis by changing the default "Steps per mm" x-axis from 94.117647 to 96.53 but if I change the y-axis to this same value, or higher there is no observable change during the print for the y-axis.

fcp.ini
`[a]
[b]
[machine]
[x]
steps_per_mm=96.53
[y]
steps_per_mm=96.53
[z]`

Seamuswg, 2018-03-03:
I am having the same issue on a Makerbot rep2 and changing the steps per mm has no effect. It was working fine for over a year and then started to print 5% bigger in the x-axis.
Any help would be appreciated.

boringuy, 2019-08-03:
I have the same problem. Both x and y doesn't seem to change much after I added 15 to both.

Snave3d, 2019-09-07:
Same here: but with extruder calibration using GPX plugin on Octoprint. I can change the steps p/mm on tool 0 and it works as expected, perfect. But if I calibrate and change steps p/mm on tool 1 it makes no difference even if I add a really high value it makes no difference at all? really odd why would it work for one and not the other?

lorenzoarna, 2024-04-05:
How can I access the steps per millimeter in the Crator Pro printer??
because I did tests with readyrface but I can't access it, I would like to understand how to do it and what to do?