markwal/OctoPrint-GPX #56: Possible Octoprint 1.3.7 Incompatibility - FFCP

BryanSmithDev, 2018-03-31:
See the comment below for some more reference:
https://github.com/foosel/OctoPrint/issues/2529#issuecomment-377633452

I have a FlashForge Creator Pro 2016 that was working perfectly under Octoprint 1.3.6. Upon upgrading to Octorpint 1.3.7rc3, I noticed that only one extruder was showing up and that my heatbed was not showing up either in Octoprint. I tried to print anyway and while it did start the print, it was loud, "grindy" and the movement while resembling what it was supposed to be doing, did not appear to be quite right.

Reverting Octoprint back to 1.3.6 did not fix the issue. Nor did uninstalling and reinstalling GPX. I assumed maybe the config had gotten messed up due to an issue with 1.3.7 and it was still reading it despite removing and reinstalling the plugin. Not until I manually deleted gpx.ini (and any other GPX related files) from the .octoprint folder, went to the GPX settings menu and hit 'save' to regenerate the gpx.ini did it start working back to normal. Keep in mind I had reverted back to Octoprint 1.3.6 at this point, which it previously worked fine on. So I simply got back to where I started before upgrading Octoprint.

Since it seemed to be related to config of GPX plugin. I saved my gpx.ini file that was working on 1.3.6. Then I upgraded again to 1.3.7rc3. It was back to not working correctly as before. I saved the 1.3.7rc3 gpx.ini. They quite different as you can see from the attachments. The 1.3.7rc3 gpx.ini has a lot more config lines most of which are mentioning things about not having a heat bed and extruders. Which is obviously one of the issues I was having.

One thing I tried was manually removing all the lines in the 1.3.7rc3 gpx.ini relating to not having extruders and a heatbed (effectively making it the same as the 1.3.6 gpx.ini). I disconnected the printer and reconnected and it seems to work fine now. However, if I change some GPX settings it regenerates all the lines I took out to get it working again. I am not sure what would cause these lines that I have never seen before in the gpx.ini to be added after an Ocotprint upgrade. Let me know if you need any logs or other info.

The one thing that throws me off is why this is affecting me but you (@markwal) said yours was working fine on rc2 of 1.3.7.

[gpx_op1.3.6.txt](https://github.com/markwal/OctoPrint-GPX/files/1865172/gpx_op1.3.6.txt)
[gpx_op1.3.7rc3.txt](https://github.com/markwal/OctoPrint-GPX/files/1865173/gpx_op1.3.7rc3.txt)

markwal, 2018-04-05:
I'm not sure what changed, but it looks like the Json parser returns something different when sent an 'undefined'. In any case, I think I may have this fixed in 2.6.1.

leeph, 2018-04-05:
I had exactly the same issue with my Qidi Tech 1 - which as I understand it is basically the same as a FlashForge Creator Pro. Replicator 1 Dual profile etc.

I wish I'd knew about removing gpx.ini - I ended up rebuilding my Octoprint machine from the ground up to get it working again.

I just tried installing the 2.6.1 GPX update but I am getting an error installing it. Other plugins install OK.

Running setup.py bdist_wheel for OctoPrint-GPX: started
Failed building wheel for OctoPrint-GPX
Running setup.py bdist_wheel for OctoPrint-GPX: finished with status 'error'

I daren't touch anything in case things break again!!

leeph, 2018-04-05:
I also have an issue with the GPX plugin whereby I can't manually extrude using Octoprint controls from the left hand nozzle (T1). It works fine for T0, but clicking the manual extrude button for T1 starts the carriage moving very slowly in the X axis, and nothing is extruded.

I believe all my settings are as correct as they can be, as everything else works great. Any ideas why this would be, Markwal?

BryanSmithDev, 2018-04-05:
@markwal 
I'm not at home but I tried to install 2.6.1 remotely from work (Don't judge). It failed to install for me as well. I can get you logs if you need them later.

adcurtin, 2018-05-17:
I'm running 2.6.2 and just started having this problem. The second extruder and hbp don't show up in the M105 command. I'm using a powerspec 3d pro, which is a flashforge creator pro clone, and OctoPrint 1.3.8 running on OctoPi 0.14.0.

adcurtin, 2018-05-17:
sweet, I found out the UI had gotten out of sync with the config file. The UI said I was set to flashforge creator pro, but it doesn't seem like I was. Changed to rep1 dual, saved, then back, and saved again, and now it's working again.

prodestrian, 2018-10-06:
My Flashforge Creator Pro 2016 just had this issue too (after an Octoprint upgrade), thanks to the suggestion @adcurtin posted I managed to get it working again. I changed the Printer from Flashforge back to Default, connected, disconnected, changed it back to Flashforge and reconnected. All good now! This is after several months of printing via SD card, it felt like the dark ages! :smile: