markwal/OctoPrint-GPX #58: Odd issues with heated bed

paukstelis, 2018-04-08:
I'm using Octoprint 1.3.6, FlashForge Creator Pro, and the most recently distributed version of OctoPrint-GPX. Prior to updating GPX everything was working as expected. After I updated I see odd behavior with the heated bed. It heats to the correct temperature before beginning the print, but just a few seconds into the first layer the bed temp sets to 130 degrees if I use the left nozzle, and turns off if I am using the right nozzle. There are no bed temperature changes in the gcode. Serial log does not show any commands being sent that would change the temperature. Only some temperature reporting, a few extrusion moves, and then the M105's show the set temperature is 130 (or 0). Anyone else seeing this?

paukstelis, 2018-04-10:
Here is the serial log spanning the unintended temperature change:

> 2018-04-09 20:19:13,836 - Recv:  T:216 /220 B:45 /45 T0:23 /0 T1:216 /220 @:0 B@:0
2018-04-09 20:19:15,978 - Recv:  T:217 /220 B:45 /45 T0:23 /0 T1:217 /220 @:0 B@:0
2018-04-09 20:19:17,015 - Recv: ok
2018-04-09 20:19:17,019 - Send: N43 G1 X-13.550 Y4.819 E0.1949*123
2018-04-09 20:19:17,027 - Recv: ok
2018-04-09 20:19:17,028 - Send: N44 G1 X-13.972 Y4.398 E0.0229*126
2018-04-09 20:19:17,035 - Recv: ok
2018-04-09 20:19:17,036 - Send: N45 G1 X-13.972 Y1.750 E0.1017*116
2018-04-09 20:19:19,140 - Recv:  T:218 /220 B:45 /45 T0:24 /0 T1:218 /220 @:0 B@:0
2018-04-09 20:19:21,278 - Recv:  T:219 /220 B:45 /45 T0:23 /0 T1:219 /220 @:0 B@:0
2018-04-09 20:19:23,428 - Recv:  T:219 /220 B:45 /45 T0:24 /0 T1:219 /220 @:0 B@:0
2018-04-09 20:19:25,567 - Recv:  T:219 /220 B:45 /45 T0:24 /0 T1:219 /220 @:0 B@:0
2018-04-09 20:19:25,702 - Recv: ok
2018-04-09 20:19:25,703 - Send: N46 M105*21
2018-04-09 20:19:25,747 - Recv: ok T:219 /220 B:45 /130 T0:24 /0 T1:219 /220 @:0 B@:0
2018-04-09 20:19:25,748 - Send: N47 G1 X-14.537 Y1.750 E0.0217*127
2018-04-09 20:19:25,760 - Recv: ok
2018-04-09 20:19:25,761 - Send: N48 G1 X-14.958 Y1.328 E0.0229*115
``

ianc1215, 2018-09-03:
Yeah, I am seeing the same issue. I have been racking my brain trying to figure out where the issue was.
The issue reproducible in my setup no matter which bed temperature I use in Cura. I had set my bed temperature to 50C so it would heat up fast. As soon as the printer finished the initial heating cycle the head moves to the start and see the LCD and Octoprint read 130C for the heated bed temperature and that when I kill the job.

Any idea what is going on?

Hobof30, 2018-10-01:
Also having this issue. Though I am very new to printing (2 years under my belt) ill try to describe my issue in as much detail:

What is programed:
left extruder 50c
right extruder 220c
bed 100c.

What happens:
left extruder starts to warm up to 100 for a few seconds then turns back off.
right extruder warms up to 220
bed remains off and print begins.

also using Octoprint 1.3.9
and gpx 2.6.3
using Qidi Tech replicator dual.

When printing file from SD card, no issues. however when printing the g code from octoprint through gpx i have these issues. octoprint controls my printer perfectly. manually set temps ect.

chuckb93644, 2018-10-21:
I have the same issues as Hobof30 only I am using a Flash Forge Creator Pro.  I am also using Octoprint 1.3.9 and gpx 2.6.3.  I purchased the Raspberry Pi on 18 June 2018.  After the standard installation, everything worked well.  After I had printed fifteen or twenty times, it stopped working.   After many frustrating attempts to geting things working again,  I installed the gpx 2.6.3 file.  I can now talk to the printer, and get the same resullts as Hobof30.  Printing files from and SD card works well with no issues.

markwal, 2020-11-17:
I believe the problems here were related to settings getting lost on upgrade.  You should be able to fix it by choosing Settings->GPX in OctoPrint and then choosing your printer again from the dropdown and then clicking Save.

closed 2020-11-17 by markwal