markwal/OctoPrint-GPX #59: how do you install an old version?

mark-hahn, 2018-04-26:
I downloaded an older version of the GPX plugin as part of an attempt to isolate a problem.  I un-tarred the file into a folder but I can't find any instructions on installing it.  Help please?

markwal, 2020-11-17:
In OctoPrint's plugin manager UI (Settings->Plugin Manager->Get More...) you can choose to install a plugin from a URL or from an uploaded file.  You want to use the entire tar ball (don't untar it).  Recently the tarball extension changed to .tgz to workaround a bug in OctoPrint, so for versions before 2.6.6 you want .tar.gz, and for recent versions you want .tgz.

For example, you can use the following url to install version 2.6.4:
https://github.com/markwal/OctoPrint-GPX/releases/download/2.6.4/OctoPrint-GPX.tar.gz

closed 2020-11-17 by markwal

mark-hahn, 2020-11-18:
Thanks for your trouble, even if my problem was 2.5 years ago.  :-)    Having the info here will be good for googlers.