markwal/OctoPrint-GPX #64: Resource Busy Error with Replicator 1

jmwright, 2018-08-22:
I've got a Replicator 1 hooked up to a Raspberry Pi running the latest Octopi image (0.15.1). The printer has the latest firmware. I've confirmed that the printer works over USB in Windows, and it shows up if I do `lsusb` or check `dmesg` after the printer is plugged in. Instead of `/dev/ttyUSB*` or `/dev/ttyAC*`, the printer ends up at `/dev/usb/lp0`. I had to put my user in the lp group so that I could access that file, and I added `/dev/usb/lp*` as a glob in the extra ports field in the Octoprint settings. Then I manually select that port and the baud rate (115200). Octoprint always says `device or resource busy` though when I try to connect to the printer. It seems like everyone else's printers show up as something like `/dev/ttyA*`, so I'm not sure if I've missed a step somehow.

rpavlik, 2019-10-28:
If you're still having this issue, what shows up in dmesg? `/dev/usb/lp0` is a USB printer (like, paper printer) or parallel-to-USB adapter - if you're really getting this from your device there's a problem, the most likely issue that's fixable would be with the USB Serial converter (atmega8u2, probably) firmware.

Did you modify any configuration files on your image or modify any firmware? I've never heard of a replicator showing up as a (paper) printer before, so this is truly a puzzle.

jmwright, 2019-10-28:
@rpavlik Thanks for the reply. I don't have access to that printer anymore. I ended up using a Taz 6 instead and it worked fine. I don't think I had modified the config files or the firmware, but it's been awhile. I'm going to go ahead and close this issue.

closed 2019-10-28 by jmwright