markwal/OctoPrint-GPX #67: fan stops in last layer

biggoldensun, 2019-01-30:
The cooling fan is switched off before the last layer is finished printing...

FSinatraisback, 2021-01-26:
Same to me. For 3D Printing no Problem, but I want to Lasercut with my CTC and so the fan gCodes M106 and M107 should start or stop the fan immediatly before or after the last movement of the printhead. 
I managed to start the fan in the right time by placing a G4 command before. To stop the fan i can't do this. The Laser will burn a hole during this delay.

markwal, 2021-02-23:
Could you save your gcode somewhere (DropBox maybe?) and share a link here?  I'd like to take a look.

FSinatraisback, 2021-02-24:
This is a GCode that works for me.

[GCode](https://files.mycloud.com/home.php?brand=webfiles&seuuid=da88842c34d62507c60ce55882a9d4a2&name=GITTest)

 I use MatterControl at the moment to Laser directly over USB to printer. Using this software i have to place a G4 P0 command although to get it work. The difference is that here the P0 means really 0 ms. Printing it over octoprint with this command i get a delay of about 100 or more ms. Like i said before, no problem for the "Laser on" command, but for "Laser off" i burn a hole with this delay.

markwal, 2021-02-24:
I suspect @biggoldensun 's issue is different.  I suspect @biggoldensun may have an M107 in the gcode that turns off the fan before he wants it off and when he posts the gcode, I'll help him figure out how to get his slicer to stop doing that.  The other possibility that I can think of is the fan confusion problem.

@FSinatraisback is an octoprint gcode latency problem.  I doubt I'll be able to fix that in the plugin itself.  For that, the most common workaround is to save x3g to an SD card and use OctoPrint merely to trigger printing of the SD card file.