markwal/OctoPrint-GPX #73: OctoPrint-GPX for macOS

michaelbaisch, 2019-11-20:
Hello everyone,

I would like to use OctoPrint with GPX with my old Thing-o-matic. But I'm getting errors while trying to install the plugin:

```
Installing plugin from https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz...
/Users/michael/OctoPrint/venv/bin/python -m pip --disable-pip-version-check install https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz --no-cache-dir
Collecting https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
Downloading https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz (125kB)
Requirement already satisfied: OctoPrint in ./venv/lib/python2.7/site-packages/OctoPrint-1.3.12-py2.7.egg (from OctoPrint-GPX==2.6.3) (1.3.12)
Requirement already satisfied: flask<0.11,>=0.10.1 in ./venv/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.10.1)
Requirement already satisfied: Jinja2<2.9,>=2.8.1 in ./venv/lib/python2.7/site-packages/Jinja2-2.8.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (2.8.1)
Requirement already satisfied: tornado==4.5.3 in ./venv/lib/python2.7/site-packages/tornado-4.5.3-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (4.5.3)
Requirement already satisfied: Flask-Login<0.3,>=0.2.11 in ./venv/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.2.11)
Requirement already satisfied: regex!=2018.11.6 in ./venv/lib/python2.7/site-packages/regex-2019.11.1-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (2019.11.1)
Requirement already satisfied: Flask-Principal<0.5,>=0.4 in ./venv/lib/python2.7/site-packages/Flask_Principal-0.4.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.4.0)
Requirement already satisfied: Flask-Babel<0.13,>=0.12 in ./venv/lib/python2.7/site-packages/Flask_Babel-0.12.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.12.2)
Requirement already satisfied: Flask-Assets<0.13,>=0.12 in ./venv/lib/python2.7/site-packages/Flask_Assets-0.12-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.12)
Requirement already satisfied: werkzeug<0.16,>=0.15.1 in ./venv/lib/python2.7/site-packages/Werkzeug-0.15.6-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.15.6)
Requirement already satisfied: cachelib<0.2,>=0.1 in ./venv/lib/python2.7/site-packages/cachelib-0.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.1)
Requirement already satisfied: PyYAML<6,>=5.1 in ./venv/lib/python2.7/site-packages/PyYAML-5.1.2-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (5.1.2)
Requirement already satisfied: markdown<3.1,>=3.0 in ./venv/lib/python2.7/site-packages/Markdown-3.0.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (3.0.1)
Requirement already satisfied: pyserial<3.5,>=3.4 in ./venv/lib/python2.7/site-packages/pyserial-3.4-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (3.4)
Requirement already satisfied: netaddr<0.8,>=0.7.19 in ./venv/lib/python2.7/site-packages/netaddr-0.7.19-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.7.19)
Requirement already satisfied: watchdog<0.10,>=0.9.0 in ./venv/lib/python2.7/site-packages/watchdog-0.9.0-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.9.0)
Requirement already satisfied: sarge==0.1.5post0 in ./venv/lib/python2.7/site-packages/sarge-0.1.5.post0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.1.5.post0)
Requirement already satisfied: netifaces<0.11,>=0.10.9 in ./venv/lib/python2.7/site-packages/netifaces-0.10.9-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.10.9)
Requirement already satisfied: pylru<1.3,>=1.2 in ./venv/lib/python2.7/site-packages/pylru-1.2.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.2.0)
Requirement already satisfied: rsa<5,>=4.0 in ./venv/lib/python2.7/site-packages/rsa-4.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (4.0)
Requirement already satisfied: pkginfo<1.6,>=1.5.0.1 in ./venv/lib/python2.7/site-packages/pkginfo-1.5.0.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.5.0.1)
Requirement already satisfied: requests<3,>=2.21.0 in ./venv/lib/python2.7/site-packages/requests-2.22.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (2.22.0)
Requirement already satisfied: semantic_version<2.7,>=2.6 in ./venv/lib/python2.7/site-packages/semantic_version-2.6.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (2.6.0)
Requirement already satisfied: psutil<5.7,>=5.6.1 in ./venv/lib/python2.7/site-packages/psutil-5.6.5-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (5.6.5)
Requirement already satisfied: Click<8,>=7 in ./venv/lib/python2.7/site-packages/Click-7.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (7.0)
Requirement already satisfied: awesome-slugify<1.7,>=1.6.5 in ./venv/lib/python2.7/site-packages/awesome_slugify-1.6.5-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.6.5)
Requirement already satisfied: feedparser<5.3,>=5.2.1 in ./venv/lib/python2.7/site-packages/feedparser-5.2.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (5.2.1)
Requirement already satisfied: chainmap<1.1,>=1.0.3 in ./venv/lib/python2.7/site-packages/chainmap-1.0.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.0.3)
Requirement already satisfied: future<0.18,>=0.17.1 in ./venv/lib/python2.7/site-packages/future-0.17.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.17.1)
Requirement already satisfied: scandir<1.11,>=1.10 in ./venv/lib/python2.7/site-packages/scandir-1.10.0-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.10.0)
Requirement already satisfied: websocket-client<0.57,>=0.56 in ./venv/lib/python2.7/site-packages/websocket_client-0.56.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.56.0)
Requirement already satisfied: wrapt<1.12,>=1.11.1 in ./venv/lib/python2.7/site-packages/wrapt-1.11.2-py2.7-macosx-10.15-x86_64.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.11.2)
Requirement already satisfied: futures<3.3,>=3.2 in ./venv/lib/python2.7/site-packages/futures-3.2.0-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (3.2.0)
Requirement already satisfied: emoji<0.6,>=0.5.1 in ./venv/lib/python2.7/site-packages/emoji-0.5.4-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.5.4)
Requirement already satisfied: monotonic<1.6,>=1.5 in ./venv/lib/python2.7/site-packages/monotonic-1.5-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.5)
Requirement already satisfied: frozendict<1.3,>=1.2 in ./venv/lib/python2.7/site-packages/frozendict-1.2-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.2)
Requirement already satisfied: sentry-sdk==0.7.7 in ./venv/lib/python2.7/site-packages/sentry_sdk-0.7.7-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (0.7.7)
Requirement already satisfied: typing<4,>=3.6.6 in ./venv/lib/python2.7/site-packages/typing-3.7.4.1-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (3.7.4.1)
Requirement already satisfied: filetype<2,>=1.0.5 in ./venv/lib/python2.7/site-packages/filetype-1.0.5-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.0.5)
Requirement already satisfied: appdirs>=1.4.0 in ./venv/lib/python2.7/site-packages/appdirs-1.4.3-py2.7.egg (from OctoPrint->OctoPrint-GPX==2.6.3) (1.4.3)
Requirement already satisfied: itsdangerous>=0.21 in ./venv/lib/python2.7/site-packages/itsdangerous-1.1.0-py2.7.egg (from flask<0.11,>=0.10.1->OctoPrint->OctoPrint-GPX==2.6.3) (1.1.0)
Requirement already satisfied: MarkupSafe in ./venv/lib/python2.7/site-packages/MarkupSafe-1.1.1-py2.7-macosx-10.15-x86_64.egg (from Jinja2<2.9,>=2.8.1->OctoPrint->OctoPrint-GPX==2.6.3) (1.1.1)
Requirement already satisfied: singledispatch in ./venv/lib/python2.7/site-packages/singledispatch-3.4.0.3-py2.7.egg (from tornado==4.5.3->OctoPrint->OctoPrint-GPX==2.6.3) (3.4.0.3)
Requirement already satisfied: certifi in ./venv/lib/python2.7/site-packages/certifi-2019.9.11-py2.7.egg (from tornado==4.5.3->OctoPrint->OctoPrint-GPX==2.6.3) (2019.9.11)
Requirement already satisfied: backports_abc>=0.4 in ./venv/lib/python2.7/site-packages/backports_abc-0.5-py2.7.egg (from tornado==4.5.3->OctoPrint->OctoPrint-GPX==2.6.3) (0.5)
Requirement already satisfied: blinker in ./venv/lib/python2.7/site-packages/blinker-1.4-py2.7.egg (from Flask-Principal<0.5,>=0.4->OctoPrint->OctoPrint-GPX==2.6.3) (1.4)
Requirement already satisfied: Babel>=2.3 in ./venv/lib/python2.7/site-packages/Babel-2.7.0-py2.7.egg (from Flask-Babel<0.13,>=0.12->OctoPrint->OctoPrint-GPX==2.6.3) (2.7.0)
[... cut ...] ed --compile --install-headers /Users/michael/OctoPrint/venv/bin/../include/site/python2.7/OctoPrint-GPX Check the logs for full command output.
Running setup.py install for OctoPrint-GPX: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output
```

I know in the `Compatibility Information` it only says linux, but since GPX is running on macOS, I believe this should be possible. Any quick ideas how to make it work?

Thanks :)

Sanerix, 2020-05-05:
Some declaration problem

download and extract archive
in src/gpxresp.c
change line 900  and 906 and remove B
....
            speed=57600;
            break;
            // TODO auto detect speed when 0?
        case 0: // 0 means default of 115200
            *baudrate=115200;
        case 115200:
            speed=115200;
....

install plugin using dir name instead of archive

Script succeded for me...

Sanerix, 2020-05-05:
Either that, or replacing the src/ files with the lates GPX results in plugin loading, though I get segmentation fault thereafter

markwal, 2020-11-17:
I don't have OctoPrint running on a Mac, but since 2.6.7 fixes that compile issue, perhaps it works now?

michaelbaisch, 2022-01-26:
Now it would be nice to be able to simply select it in OctoPrint - https://github.com/markwal/plugins.octoprint.org/pull/1