markwal/OctoPrint-GPX #75: Flashforge Creator Pro will not successfully print with both extruders

gadg3ts, 2020-04-28:
Hi,
I have a new Flashforge Creator pro (sailfish 7.8) that for whatever reason won't complete a print if attempting to use both extruders via Octoprint + the GPX plugin.
However, exporting an x3g file from Cura to print from the SD card works fine.

I'm using the following:
Cura 4.6 with the FFCP profile from https://github.com/eugr/Flashforge-for-Cura - gcode flavour is set to RepRap
Octoprint 1.4.0 - Gcode flavour in the GPX plugin also set to RepRap

The issues are that either Octoprint drops the serial connection to the printer partway through, or it keeps sending the commands but the printer stops doing anything. Or sometimes it has just used only the left extruder for a 2-colour print.
(This is actually my 5th printer, but the first with a non-Marlin firmware.)
I've attached the octoprint & gpx plugin logs after a fresh start (and removing the previous logfiles)
Any assistance would be appreciated.
Thanks,
Sean

[FCPRO_10mm Cube Halves.gcode.txt](https://github.com/markwal/OctoPrint-GPX/files/4547548/FCPRO_10mm.Cube.Halves.gcode.txt)
Also the gcode of the split 10mm cube I'm using as a test.
[octoprint.txt](https://github.com/markwal/OctoPrint-GPX/files/4547529/octoprint.txt)
[plugin_GPX.txt](https://github.com/markwal/OctoPrint-GPX/files/4547531/plugin_GPX.txt)

markwal, 2020-05-10:
It looks like it would work on my FFCP sailfish 7.8.  What are your GPX settings in OctoPrint.  Is it set to RepRap flavor as well?  I suspect it is set to "MakerBot" because the GPX project from way back has built into it that a T0/T1 is not sticky and has to be preceeded by an M135 to switch extruders.  The gcode you have from Cura expects T0 to switch to the right extruder and T1 to switch to the left.  I think that won't work if your gcode flavor is set to "MakerBot" in OctoPrint.

gadg3ts, 2020-05-10:
Right. That's helpful :)
Cura & the GPX plugin are both set to RepRap.
However, given what you've said, I generated a new gcode file from Cura and that, even though it's set to RepRap, doesn't add 
M135 T0 
just 
T0
for selecting the extruders.
so I've added in the Machine settings -> Extruder settings in Cura 
M135 Tx (where x = 0 or 1)

although, this does seem to add them like the below. (I'll give it a go and see what happens)
T1
M82 ;absolute extrusion mode
M135 T1
M83 ;relative extrusion mode

gadg3ts, 2020-05-10:
OK, this seems to work as expected now.
I've successfully printed (via octoprint) a (near enough) 10mm cube with two colours.
Although it did pause about 71% with the octoprint terminal telling me to "press unpause on the LCD".
Thanks for your help.