markwal/OctoPrint-GPX #76: UnicodeDecodeError

sh7777, 2020-05-12:
Hi, I am getting the following error :
State: Offline (Error UnicodeDecodeError: ''ascii' codec can't decode byte 0xf6 in position 13: ordinal not in range(128)' @ comm.py:_readline:2823) 

Any idea what the problem could be? I have a FFCP connected via USB. 

Thanks

markwal, 2020-05-13:
It could be the contents of the file you are printing?  If so, please put somewhere and share a link here.  If you weren't printing, what action did you take?  Connect?

markwal, 2020-05-13:
The most common cause is that you're trying to print an x3g through OctoPrint.  GPX translates from gcode to x3g on the fly, so to print from OctoPrint, you want to print gcode.  To print from the SD card in the printer, you want x3g.

sh7777, 2020-05-13:
Hi. I only got this error after i applied the latest update when trying to connect to my printer. 
I wiped and reinstalled octoprint image on my Raspberry Pi and reinstalled GPX but still get this error when trying to connect.

markwal, 2020-05-13:
I may need verbose logs, but first, what printer?

markwal, 2020-05-14:
I pushed a new release 2.6.5 that may fix this.  I'm guessing here, but perhaps changes to GPX and OctoPrint for Python 2/3 caused this.

brassbell, 2020-05-17:
Unexpected error while reading serial port, please consult octoprint.log for details: UnicodeDecodeError: ''ascii' codec can't decode byte 0xeb in position 11: ordinal not in range(128)' @ comm.py:_readline:2823
Changing monitoring state from "Operational" to "Offline (Error: UnicodeDecodeError: ''ascii' codec can't decode byte 0xeb in position 11: ordinal not in range(128)' @ comm.py:_readline:2823)"
Connection closed, closing down monitor, Looks like the problem still exists on flash forge creator pro on sailfish 7.8

markwal, 2020-05-17:
But I have a Flashforge Creator Pro with Sailfish 7.8 and it doesn't occur.  Perhaps the GPX plugin gets disabled on upgrade for some reason?  You can tell by the GPX panel disappearing from settings.  Also octoprint.log would be helpful.

spagatticode, 2020-05-20:
I got the same Error here after Update to GPX 2.6.5 with FF Creator Pro.
Problem was a File on the SD-Card with an Filename conatining a non-ASCII character.
The directory structure will be read with gcode "M20" while connecting.

Quick Workaround: Remove SD-Card.

On my SD-Card there are some Manuals for the Creator-Pro with Chinese Characters in Filename, I think they are provided by FlashForge by default on the SD-Card shipped with the Creator Pro.

markwal, 2020-05-20:
Aha!  Thank you for the info. I'll be able to reproduce and fix that.

markwal, 2020-05-20:
Shoot.  It doesn't reproduce the error for me.  The code I put in 2.6.5 to filter out non-ascii characters effectively filters out the non-ascii characters for me.  @spagatticode could you copy and paste the non-ASCII filenames you've got on your SD card here?  I could use those to test.

spagatticode, 2020-05-21:
I can do that, but I'm not in office until next monday. 
I dont have access to the printer before.
I will send you the filenames and the log output then.

spagatticode, 2020-05-27:
Hi @markwal,
originally my SD-Card looked like this:
[image]

As I renamed the files with the chinese characters already, I reproduced the Bug today by just putting an empty File named `快速启动指南.txt` on the Card (SDHC 8Gb, FAT32). 
This produces the following log during connection:

```
Recv: start
Send: N0 M110 N0*125
Recv:  Sailfish v7.8
Recv: echo: gcode to x3g translation by GPX
Send: N0 M110 N0*125
Recv: SD card ok
Recv:  T:26 /0 B:16 /0 T0:26 /0 T1:26 /0 @:0 B@:0
Recv:  T:26 /0 B:16 /0 T0:26 /0 T1:26 /0 @:0 B@:0
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Connecting" to "Operational"
Recv: ok
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok PROTOCOL_VERSION:0.1 FIRMWARE_NAME:Sailfish FIRMWARE_VERSION:7.8 FIRMWARE_URL:http://www.sailfishfirmware.com MACHINE_TYPE:r1d EXTRUDER_COUNT:2
Send: N2 M20*19
Recv: ok
Recv: Begin file list
Recv: ENG
Recv: ENG
Send: M105
Recv: ok T:26 /0 B:16 /0 T0:26 /0 T1:26 /0 @:0 B@:0
Recv: flashprint_3.14.0_i386.deb
Recv: FlashPrint_3.14.0_mac.zip
Recv: FlashPrint_3.14.0_win32.zip
Recv: FlashPrint_3.14.0_win64.zip
Recv: flashprint_3.14.0_amd64.deb
Recv: Projekte
Recv: Creator Pro Quickstart CN.pdf
Unexpected error while reading serial port, please consult octoprint.log for details: UnicodeDecodeError: ''ascii' codec can't decode byte 0xeb in position 0: ordinal not in range(128)' @ comm.py:_readline:2823
Changing monitoring state from "Operational" to "Offline (Error: UnicodeDecodeError: ''ascii' codec can't decode byte 0xeb in position 0: ordinal not in range(128)' @ comm.py:_readline:2823)"
Connection closed, closing down monitor
```

theadamfarah, 2020-06-09:
FWIW I had this exact same issue. Brand install on 3B+; Brand new FFCP.

I was only trying to connect, but kept coming up against the exact same message.

I installed my only two updates, which were GPX and v2.6.5. Still no dice.

The simple fix was to unseat the SD card.

markwal, 2020-06-13:
Just occurred to me that I might not be seeing this problem because I use a FlashAir SD card.  Maybe its doing something to the filename in firmware...  Just  a conjecture.  I'll see if I can find another SD card to test with.

markwal, 2020-06-13:
OK.  I have a repro now with @spagatticode 's most recently posted filename.  Should be able to fix now.

markwal, 2020-06-14:
I believe this is fixed in 2.6.6, hopefully

closed 2020-06-14 by markwal

spagatticode, 2020-06-15:
FYI: 2.6.6 has fixed this for me! :thumbsup:

markwal, 2020-06-15:
Thank you @spagatticode for finding the repro and for verifying the fix.

I should point out though, that the fix keeps it from disconnecting immediately on connection, it doesn't allow you to actually print SD files with non-ASCII filenames. 😟