markwal/OctoPrint-GPX #77: Cancelling a Print causes hang up

leitenky, 2020-05-14:
When you cancel a print using octoprint, octoprint gets stuck on cancelling (even though the printer has stopped printing) and requires you to disconnect and reconnect the printer through octoprint. Note if you cancel the print on the printer LCD the job cancels successfully.

hbh7, 2020-05-14:
Same on Qidi Tech 1

leitenky, 2020-05-15:
Lets home the author of the plug-in can help us out! Im willing to test if needed

Joshi425, 2020-05-15:
can you post a serial.log from a cancel? Please beware that cancelling during long running commands like heating up can take some time.

leitenky, 2020-05-15:
I selected Cancel from withing Octoprint on a print job. the Printer stopped moving a couple seconds later and completely cooled but octoprint stayed in a "Cancelling" mode. Disconnecting and reconnecting (through Ocotoprint) seems to be the only way a print will go from cancelling to operational again.

[image]

[serial (1).log](https://github.com/markwal/OctoPrint-GPX/files/4634603/serial.1.log)

leitenky, 2020-05-15:
Here is a Log of what happens if I cancell from the Printer LCD. When I do this, octoprint goes back to operational after a few seconds without disconnecting
[serial.log](https://github.com/markwal/OctoPrint-GPX/files/4634667/serial.log)

leitenky, 2020-05-21:
> can you post a serial.log from a cancel? Please beware that cancelling during long running commands like heating up can take some time.

Did you have a chance to look into this? It's there anything else I can provide?

markwal, 2020-06-14:
It looks to me like it is that M108 that is causing problems.  It's waiting for the temperature to reach 0 which it never will.  Do you have "M108" in one of the gcode fields in OctoPrint settings?  A fix would be for me to fix M108 to not wait if there is no target set.  A workaround in the meantime would be for you to remove the M108 from the gcode script:

[image]

leitenky, 2020-06-15:
I can check my PostScript to see if it's still there. But if it's to cancel heating how would the heaters know to turn off without that command?

markwal, 2020-06-15:
M108 isn't cancel heating. It's wait for temp. You want M104 S0 and M140 S0 to cancel heating.

leitenky, 2020-06-15:
I just checked my scripts in simplify3d and my octoprint gcode scripts and I don't see m108 anywhere.

markwal, 2020-06-15:
Thank you for checking. M108 is the preemptive cancel code.  Your log shows there is still a race happening between the internal state of GPX and my plugin.  Hmm...

leitenky, 2020-06-15:
No problem, Thanks for looking into the issue! If there is anything else I can provide let me know

leitenky, 2020-07-01:
Hey just wondering if you had any luck with the issue?

markwal, 2020-07-01:
Yes. I believe I have made some progress with making a timeout to help force resync after cancel, but unfortunately can continue my experimentation until July 6th.  Away from my printers atm.

DaveTSG, 2020-10-25:
Hi @markwal. As per [the above](https://github.com/OctoPrint/OctoPrint/issues/3784), I'm also having this issue (Flashforge Creator Pro). Do you need any logs in particular or need me to raise a separate issue? Thanks

leitenky, 2020-10-28:
Yes I still have the issue to so if there's something I can provide again please let me know

adamdoyle, 2020-10-29:
@markwal: If you need any additional logs or want to run any tests, I'm also happy to help.  I have the same issue cancelling with my FFCP via GPX plugin.

markwal, 2020-11-10:
OK.  This turns out to be my plugin not keeping up with some great work on the OctoPrint side.  OctoPrint will now force send an M108 by default on cancel.  We've already got code in GPX which will see that M108 and cancel heatups and clear the queue.  However, some code I added way back before OctoPrint had this feature will also do this when it sees a cancel event.  I did this because I didn't want to wait for the heatup to complete and OctoPrint won't send anything until then (or didn't at the time).  Now, this causes a problem because GPX sees two cancels, but only one of them gets the sync (@clear_cancel).

So the plugin is now unnecessarily complicated for cancel and probably things will work better if I just delete that old code and rely on OctoPrint to do the force send of M108.  Testing that now.

markwal, 2020-11-17:
I've attempted a fix in release 2.6.7.  It appears to work for me, but since it is a bit of a race problem between multiple threads there is still the possibility that I haven't really got them all. But fingers crossed, it should be less common now.

closed 2020-11-17 by markwal