markwal/OctoPrint-GPX #80: Cannot connect to Flash Forge Creator since last update

BenBtg, 2020-05-31:
After a recent update I get the error connecting to my Flash Forge Creator.
Looks to be related to a recent change as the error log includes the very line that was modified recently.

" File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GPX/gpxprinter.py", line 190, in readline
    return self.readline_str().encode('ascii', errors='ignore')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 14: ordinal not in range(128)
2020-05-31 10:06:10,317 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline (Error: UnicodeDecodeError: ''ascii' codec can't decode byte 0xf6 in position 14: ordinal not in range(128)' @ comm.py:_readline:2823)""

BenBtg, 2020-06-01:
Looks to be related to #76 . 
Removing SD card works as a workaround. Must be issue parsing characters on SD card.

markwal, 2020-06-13:
Correct.  I think this is the same as #76

closed 2020-06-13 by markwal