markwal/OctoPrint-GPX #81: FFCP Temperature Drop after last Update

prodestrian, 2020-06-06:
Hi,
I have a Flashforge Creator Pro 2016, I've been running it via OctoPrint for several years now via the GPX plugin, always sliced via Simplify3D.

Tonight I performed an update to the latest version of GPX, and now I'm experiencing temperature drops.

Scenario:
1. I heat the printer to 220C (using either OctoPrint's UI or the printer LCD screen)
1. It stays stable at this temperature
1. I upload a sliced gcode file with the extruder temperature also set to 220C
1. I start the print via the OctoPrint UI
1. Shortly after the print starts, at the exact moment when it would start heating the extruder, the temperature drops all the way down to 195C
1. It continues to try to get back up to 220C, but never gets above around 200C

You can see it clearly in this screenshot. In this example my printer was only manually preheated to 215C, but instead of climbing to 220C it immediately dropped.
[image]

I couldn't tell if this was related to other recent issues (#76 or #80 ) but it didn't sound like the same problem.

To get a print going tonight, I preheated to 220C, started the print, and then when the temperature dropped to 195 I used the printer control panel to change the temperature back up to 215C (bypassing whatever commands OctoPrint has sent).
This workaround is working so far:
[image]

I haven't tried slicing directly to a 3GX file in S3D and printing from the SD Card, but I'll try that next when (if) this print finishes (just to be absolutely sure it's OctoPrint related). I also haven't tried printing in the other extruder yet. I'm printing ABS, 200C is too low. My left extruder (T1) is currently loaded with PLA.

The only thing which has changed between now and the last time I printed ABS a few weeks ago was that I applied updates to the GPX and RequestSpinner plugins, nothing else has been updated or modified.

Please let me know if you need more information from me and I can try to post it here.

Thanks

markwal, 2020-06-13:
It could be that your GPX settings got reset.  It sounds like the flavor is set the wrong way.  Check the GPX settings panel in OctoPrint.  What flavor do you usually use?  My most frequent test set up is using Cura -> RepRap flavor, so perhaps this is a problem with MakerBot flavor?

prodestrian, 2020-07-04:
Hey, I use the Makerbot flavor. My slicer is Simplify3D. I always export to "Standard G-Code".

G-Code file attached:
[7mm_tube.zip](https://github.com/markwal/OctoPrint-GPX/files/4871948/7mm_tube.zip)

As far as I can tell nothing has changed in the GPX settings:
[image]

I'm open to swapping to RepRap flavor, but I don't know how well that works with S-3D.

prodestrian, 2020-07-04:
My workaround at the moment is to set my print temperature 5 degrees cooler than I want it, then once the print starts (and the temperature plummets) I just use the front control panel to bump it up 5 degrees to where I want it. After that it seems to stay there comfortably for the whole print.

prodestrian, 2020-07-07:
Well, this is weird. It's working fine now. I did do a couple of updates and restart, but I'm a few minutes into the print and the temperature has been completely stable! Same settings that I posted above.
Currently running OctoPrint 1.4.0 with GPX 2.6.6 (not sure what updated though, I'm running 6 or 7 plugins).

I might do a few more prints, then if everything is still working I'll close the issue :+1: