markwal/OctoPrint-GPX #82: Makerbot G code Beep commands

InertiaImpact, 2020-06-10:
OctoPi with a Makerbot Replicator 2X

I am trying to have it beep, however it doesn't appear to respond to the M300 command ( M300 S300 P1000 ) from the terminal or from within Gcode file.

I have also tried M42 which gets the unrecognized response.

I will be happy to try anything you need. Thanks!

markwal, 2020-06-13:
The original GPX translator doesn't have any code for M42.  It does however attempt to translate M300 into an x3g frame of (8-bits)147, (16-bits)frequency, (16-bits)milliseconds, (8-bits)0.  I haven't ever tried it myself though.

markwal, 2020-06-13:
M300 by itself is the same as M300 S300 P1000 since the default frequency and milliseconds are 300 and 1000 respectively.  GPX does translate it and send it to the bot, the bot receives it and returns a positive response, but no sound emerges.  Though my test bot is a FlashForge Dual (Rep 1D clone) running Sailfish, so I don't know if it implements it other than to succeed the protocol.