markwal/OctoPrint-GPX #83: Nozzle Diameter

leitenky, 2020-06-15:
Just wondering if you can shed some light on the purpose of the nozzle diameter setting in GPX? Currently I have a dual extruder set up with 2 different nozzles but there is no way to configure GPX for that kind of setup

markwal, 2020-06-15:
GPX can recalculate the extrusion amount on the fly using that setting and some gcode "macros". The intended use is to allow the same gcode be used for different filaments and flow rates and mid print filament changes.  If you aren't using those features, the setting is ignored.

The Octoprint setting is used for slicing using a slicer plugin. I'm not sure if there is a slicer plugin that can handle two different filament widths for a dual setup or not. Either way, that's not particular to the GPX plugin.

leitenky, 2020-06-15:
Thanks for the info and the great plugin

closed 2020-07-01 by leitenky