markwal/OctoPrint-GPX #85: OctoPrint 1.4.1 - Can Not Install Plugin

sirhakon, 2020-08-04:
When trying to install the Plugin I get the following error:
```
Installing plugin "GPX" from https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz...
Error!
Could not install plugin from https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz, was neither a plugin archive nor a single file plugin
```
This is with a fresh install of OctoPi Version 0.17.0, running on Raspberry Pi 3 Model B Rev 1.2 - OctoPrint 1.4.1. With a fresh install of OctoPi 0.17.0 it installed fine, and I think it was running OctoPrint 1.3.12.

Simmons928, 2020-08-05:
I have a System running Octoprint 1.4.0 and I was able to install it. 
My new system running 1.4.1 is not installing.

sirhakon, 2020-08-05:
I reverted to OctoPrint 1.4.0, by following this site: [https://community.octoprint.org/t/how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image/205](url) and was able to install it again.

Cornegru, 2020-08-06:
Same problem, since I have updated Octoprint.

TheZackCodec, 2020-08-11:
Same for 1.4.2

kma-johan, 2020-08-14:
Same here. 1.4.2

jasonkuehl, 2020-08-25:
Also running Version 1.4.2 using the base OctoPrint image. This happens on a fresh installation.

cp2004, 2020-08-27:
To everyone in this thread: This is a bug in OctoPrint. Tracked in https://github.com/OctoPrint/OctoPrint/issues/3698 and solved with PR https://github.com/OctoPrint/OctoPrint/pull/3707 - therefore should make it into the next release (scheduled as 1.5.0)

# Workaround for OctoPrint 1.4.2

Download the archive to your machine by going to:
https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
Rename the file extension to `.tgz` (eg. `OctoPrint-GPX.tgz`)
Head to the plugin manager > get more > ...from archive then upload the renamed file

### Bug explanation (if anyone is interested)
`tar.gz` was not recognized properly, as it was thought as `.gz` which was not in the list of supported archives.

sirhakon, 2020-09-02:
Updated to OctoPrint 1.4.2 and followed your instructions and it installed correctly. Thank-you cp2004.

closed 2020-09-02 by sirhakon

jasonkuehl, 2020-09-02:
This is great news, thank you @sirhakon !