markwal/OctoPrint-GPX #86: Can't install GPX plugin

biggerdummy, 2020-09-03:
When trying to install GPX plugin I get the following error: Error!Could not install plugin from https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz, was neither a plugin archive nor a single file plugin

I've tried older versions of the plugin, older versions of octoprint and three different pi boards and the results are the same whether I try to install from the url or the plugin manager.

rasikrodri, 2020-09-03:
I am experiencing the same problem
[image]

ronaldmanganaro, 2020-09-07:
Yea I had the same issue. For now you can still ssh into your octopi and run sudo ~/oprint/bin/pip install "https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz"

rasikrodri, 2020-09-07:
Thank you, I will try that tomorrow

ronaldmanganaro, 2020-09-10:
Did not realize this, but the plugin still is not working for me. When I upload gcode it does not get converted to .g3x.

biggerdummy, 2020-09-10:
I did successfully use sudo to install but the plugin still won't connect. Plus, it's 2.5.6 so it constantly tells me there is an update that i need to install... which still fails to install when i run it. for now I've ignored it while i try to get this thing to connect. 

Terminal Window results:
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyACM0, baudrate 115200 from hook GPX: IOError: '[Errno 2] No such file or directory: u'/home/pi/.octoprint/data/GPX/gpx.ini'' @ comm.py:_open_serial:2784

sirhakon, 2020-09-10:
It's a bug in OctoPrint. 

**Workaround for OctoPrint**
Download the archive to your machine by going to:
https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
Rename the file extension to .tgz (eg. OctoPrint-GPX.tgz)
Head to the plugin manager > get more > ...from archive then upload the renamed file

Check here for more info: https://github.com/markwal/OctoPrint-GPX/issues/85

Austin3dPrinting, 2020-09-11:
Running the following command from @ronaldmanganaro worked great, thank you! I noticed this issue after upgrading my octoprint to Python 3

sudo ~/oprint/bin/pip install "https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz"

cp2004, 2020-09-12:
As posted by many in this thread, is a bug in OctoPrint's detection of `tar.gz` as an archive folder. Works when it is renamed as `.tgz`
Reported to OctoPrint as https://github.com/OctoPrint/OctoPrint/issues/3698, solved in the next release.

From https://github.com/markwal/OctoPrint-GPX/issues/85#issuecomment-682200814
# Workaround for OctoPrint 1.4.2

Download the archive to your machine by going to:
https://markwal.github.io/octoprint-download/OctoPrint-GPX.tar.gz
Rename the file extension to `.tgz` (eg. `OctoPrint-GPX.tgz`)
Head to the plugin manager > get more > ...from archive then upload the renamed file

### Bug explanation (if anyone is interested)
`tar.gz` was not recognized properly, as it was thought as `.gz` which was not in the list of supported archives.

markwal, 2020-11-17:
I've renamed the archive to work around this issue. Also, it appears to be fixed in the RC of the next version of OctoPrint

closed 2020-11-17 by markwal