markwal/OctoPrint-GPX #88: Never stops cancelling

DaAwesomeP, 2020-10-30:
Hello! When printing an X3G from SD card and then cancelling the print, I have to disconnect OctoPrint and reconnect it to regain control of the printer. Also, the print just stops in its tracks. It does not lower the bed and re-home X/Y.
```
Send: M27
Recv: ok
Recv: SD printing byte on line 0/0 X:-32.86 Y:-24.64 Z:0.20 E:0.00
Send: M27
Recv: ok
Recv: SD printing byte on line 0/0 X:-13.56 Y:-40.60 Z:0.20 E:0.00
Changing monitoring state from "Printing from SD" to "Cancelling"
Send: M108
Recv: ok
Recv:  T:35 /0 B:64 /0 T0:35 /0 T1:192 /0 @:0 B@:0
Send: M25
Recv: // echo: Waiting for unpause button on the LCD panel
Recv:  T:35 /0 B:64 /0 T0:35 /0 T1:192 /0 @:0 B@:0
Recv: // echo: Waiting for unpause button on the LCD panel
Recv: // echo: Waiting for unpause button on the LCD panel
```

DaAwesomeP, 2020-10-30:
I am sometimes not able to reconnect without a power cycle. This is on a FlashForge creator running Sailfish v7.7 r01591.

markwal, 2020-11-02:
What does the printer's LCD say when the terminal window says "Waiting for unpause button on the LCD panel"?

DaAwesomeP, 2020-11-02:
I'll double check again today, but IIRC it is just on the main menu (done cancelling).

thebeline, 2021-02-24:
I hate to say this, but this is still an issue.  Current workaround is to restart OctoPrint...  

Sailfish 4.7, latest OctoPrint, the LCD does go back to home, but OctoPrint does not come out of `Canceling`.

markwal, 2021-02-24:
@thebeline cancelling the print from octoprint's UI or the printer's LCD menu?

markwal, 2021-02-24:
@thebeline And are you seeing "Waiting for unpause button on the LCD panel" in the terminal window?

thebeline, 2021-02-25:
Canceling from the printer LCD leaves OP in a correct state (goes back to `operational`).

Canceling from OP cancels the print, and sets the LCD to the correct state, but hangs on `canceling`.

fridayshoes, 2021-11-11:
This is still a problem for me

me0262, 2023-08-31:
This is still an issue here.
I print directly over USB, and it goes from "Printing to Canceling".
I end up doing a full system restart.
I'll be enabling log capture and send it up if I need to cancel the print again.

```
Send: N91880 G1 X-19.693 Y-25.877 E0.04317*96
Recv: ok
Send: N91881 G1 X-19.364 Y-24.723 E0.03991*96
Recv: ok
Changing monitoring state from "Printing" to "Cancelling"
Send: N91882 M108*32
Recv: ok
Send: N91883 G1 X-18.435 Y-23.782 E0.04398*104
Recv:  T:216 /0 B:59 /0 T0:48 /0 T1:216 /0 @:0 B@:0
Recv:  T:216 /0 B:59 /0 T0:48 /0 T1:216 /0 @:0 B@:0
Recv:  T:216 /0 B:59 /0 T0:48 /0 T1:216 /0 @:0 B@:0
```

46cv8, 2023-08-31:
> I print directly over USB, and it goes from "Printing to Canceling", and stays there until I end up doing a full system restart

I have the exact same issue, after applying the latest update when I last used it 6 months ago.

me0262, 2023-08-31:
Cancelled a print (I'm having filament feed problems on some cardboard spools), and once again I'm forever canceling.
The serial.log is quite long, so I'll give the relevant snippet.
```
2023-08-31 09:20:18,542 - Send: N19875 G1 X22.55 Y-25.496 E0.10497*122
2023-08-31 09:20:18,693 - Recv: ok
2023-08-31 09:20:18,697 - Send: N19876 M105*38
2023-08-31 09:20:18,741 - Recv: ok T:215 /215 B:60 /60 T0:48 /0 T1:215 /215 @:0 B@:0
2023-08-31 09:20:18,762 - Send: N19877 G1 X20.326 Y-23.146 E0.10761*73
2023-08-31 09:20:18,793 - Recv: ok
2023-08-31 09:20:18,796 - Send: N19878 G1 X19.473 Y-20.226 E0.10118*69
2023-08-31 09:20:18,811 - Recv: ok
2023-08-31 09:20:18,814 - Send: N19879 G0 F7200 X19.057 Y-20.226*102
2023-08-31 09:20:18,895 - Recv: ok
2023-08-31 09:20:18,904 - Send: N19880 G1 F3600 X19.967 Y-17.108 E0.10803*33
2023-08-31 09:20:18,907 - Changing monitoring state from "Printing" to "Cancelling"
2023-08-31 09:20:18,920 - Recv: ok
2023-08-31 09:20:18,926 - Send: N19881 M108*35
2023-08-31 09:20:18,975 - Recv: ok
2023-08-31 09:20:19,188 - Send: (@clear_cancel)
2023-08-31 09:20:19,233 - Recv:  T:0 /0 B:0 /0 T0:0 /0 T1:0 /0 @:0 B@:0
2023-08-31 09:20:21,275 - Recv: // echo: tio->waiting = 0x80
2023-08-31 09:20:21,276 - Recv:  T:215 /0 B:59 /0 T0:48 /0 T1:215 /0 @:0 B@:0
2023-08-31 09:20:23,310 - Recv: // echo: tio->waiting = 0x80
2023-08-31 09:20:23,311 - Recv:  T:216 /0 B:59 /0 T0:48 /0 T1:216 /0 @:0 B@:0
2023-08-31 09:20:25,346 - Recv: // echo: tio->waiting = 0x80
```

and the relevant snippet from the plugin_GPX.log
```
gcode_line: N19880 G1 F3600 X19.967 Y-17.108 E0.10803*33
port_handler write: 35
d5 20 9b 57 07 00 00 b6 f9 ff ff 78 0f 00 00 00 00 00 00 f6 ff ff ff 24 15 00 00 18 8b e0 4f 40 00 0f 41 
port_handler read:
d5 d5 81 d2 
gpx_write_string_core rval = 0
gpx_return_translation rval = 0
gcode_line: N19881 M108*35
port_handler write: 4
d5 01 07 83 
port_handler read:
d5 d5 81 d2 
gpx_write_string_core rval = 0
gpx_return_translation rval = 0
waiting in gpx_write_string
gcode_line: (@clear_cancel)
translate_result: @clear_cancel
clear cancelPending
gpx_write_string_core rval = 0
leaving gpx_write_string_core 160
implicit M105
gcode_line: M105
port_handler write: 4
d5 01 18 5f 
port_handler read:
d5 d5 81 05 00 2d 00 00 00 00 00 00 00 00 12 
(line 0) Build status: build sleeping, 0 hours, 45 minutes
port_handler write: 7
d5 04 0a 00 02 00 8a 
port_handler read:
d5 d5 81 00 00 c9 
Extruder T0 temperature: 0c
port_handler write: 7
d5 04 0a 00 20 00 da 
port_handler read:
d5 d5 81 00 00 c9 
Extruder T0 target temperature: 0c
port_handler write: 7
d5 04 0a 01 02 00 21 
port_handler read:
d5 d5 81 00 00 c9 
[... 149 lines cut ...]
port_handler read:
d5 d5 81 00 00 c9 
Extruder T1 target temperature: 0c
port_handler write: 7
d5 04 0a 00 1e 00 2b 
port_handler read:
d5 d5 81 3b 00 c7 
Build platform T0 temperature: 59c
port_handler write: 7
d5 04 0a 00 21 00 1e 
port_handler read:
d5 d5 81 00 00 c9 
Build platform T0 target temperature: 0c
gpx_write_string_core rval = 0
leaving gpx_write_string_core 128
gpx_return_translation rval = 0
itio->waiting = 128
port_handler write: 4
d5 01 18 5f 
port_handler read:
d5 d5 81 05 00 2d 00 00 00 00 00 00 00 00 12 
(line 0) Build status: build sleeping, 0 hours, 45 minutes
tio->waiting = 128 and rval = 0
owaiting in gpx_write_string
```